UnchartedBull/OctoDash #687: Add Custom Actions on Sleeping Screen... like control Enclosure Rele or launch gcode command

alienboyxp, 2020-05-27:
**Additional context**
Could be interesting to add some actions/buttons available at "Sleep" screen to, for example:

- launch gcode or API custom call to Octoprint
- make available some Enclosure actions like power on or off specific rele, like to power on the printer, led strip, etc.

UnchartedBull, 2020-05-27:
GCode won't work in the sleep screen, since there is no printer connected, who could handle that. I think the API doesn't even accept the command. Custom API call will work yes, but will be a relatively large thing to implement.

Powering the printer on and off is already support with the PSUControl Plugin. Regarding the Enclosure Plugin please read #668. Feel free to open a PR yourself! It probably will take some time for me to come back to this ...

alienboyxp, 2020-05-28:
> GCode won't work in the sleep screen, since there is no printer connected, who could handle that. I think the API doesn't even accept the command. Custom API call will work yes, but will be a [...]

@UnchartedBull  thanks for your quick answer!!!
Related to send gcode I mean this functionality from Enclosure you can use/manage the plugin with custom gcodes https://github.com/vitormhenrique/OctoPrint-Enclosure/wiki/G-CODE-Control but not sure if, as you mentioned, will be available send ENC O2 S1 when printer is not connected and then the solution will be to launch API call to Enclosure plugin.

I will review the PSUControl as you mentioned but the question is if this feature to power on the printer is available at sleep screen, in any case and for my special case is not useful as I manage the power on off by GPIO rele and Enclosure is most appropriate for this scenario as provide much more features than PSU Control.

As you start to integrate Enclosure for chamber temperature will be a great feature to extend Enclosure options to manage reles or reles and, in some cases allow some actions from sleep screen.

UnchartedBull, 2020-05-29:
Yes you can set that up in the settings of OctoDash. This will then tell PSUControl to turn on the printer, if you want to exit from sleep mode.

PR #1299 (2021-01-12): Feature/custom actions on sleep
Closes #687
closed 2021-01-12 by UnchartedBull