UnchartedBull/OctoDash #693: Print history or re-print last job

ruralaccent, 2020-05-29:
I use my printers to make a lot of identical items. I often print the same job one after another. When I used the web UI of octoprint, I just clicked 'print', which then printed the same file again. 

With Octodash, the main screen resets to 'normal' and then you have to go back into the file menus to find the file and select it to print it again. Is is possible for Octodash to do the same as octoprint web UI and have n option to reprint the last job, please? 

Alternatively, perhaps there could be an easy ot access screen that shows you the last, say, 5 print jobs, so you can easily re-print one without having to wade through the print files and dirs?

Just ideas! Thanks again for the awesome plugin :)

UnchartedBull, 2020-05-29:
Will consider this for v3. I had the same situation occur to me a couple of times too.

Thanks! :)

ruralaccent, 2020-05-30:
Perhaps there could be some integration with this plugin?

https://github.com/OllisGit/OctoPrint-PrintJobHistory

UnchartedBull, 2021-05-03:
Based on #1750 this should also include a print summary at the end of a print, which should house a "print again" button.

bolsoncerrado, 2022-11-14:
Would love a PRINT AGAIN button as well!!