UnchartedBull/OctoDash #697: Would it be possible to add Tasmota Plugin support?

DolfLund, 2020-05-31:
The tasmota plugin inside Octoprint allows for a power on/off button and sleep settings of the whole printer (via smart plug).  It would be really wonderful if Octodash could allow for a power button to be added to the main screen.  I had thought this was what PSU support meant but it seems its a different type of function.

UnchartedBull, 2020-05-31:
Possible yes, but I most likely won't work on it. PSUControl does exactly what you want. You can tell OctoDash to turn on your printer, if you touch the screen while in sleep mode and you can also add some custom actions https://github.com/UnchartedBull/OctoDash/wiki/Custom-Actions#predefined-actions to turn off or on the power supply.

If you want Tasmota Plugin support feel free to open a PR. If you're fine with using PSUControl please close this issue. Thanks!

loubrucker, 2020-06-01:
I have not yet adjusted for it, but found tasmota/mqtt config guidance for PSU Control and will adapt that to achieve on/off.  It's the functionality that was my goal, it also seems that PSU Control is a better  implemented into Octoprint.

Thanks so much for the quick response, Looking forward to more of your work, loving it!

DolfLund, 2020-06-01:
I have not yet adjusted for it, but found tasmota/mqtt config guidance for PSU Control and will adapt that to achieve on/off.  It's the functionality that was my goal, it also seems that PSU Control is a better  implemented into Octoprint.

Thanks so much for the quick response, Looking forward to more of your work, loving it!

closed 2020-06-01 by DolfLund

jneilliii, 2020-08-03:
There are the same API methods in my power plugins that there are in PSU control...just requires a little additional info (ip and sometimes index number).

lightmaster, 2021-03-11:
> You can tell OctoDash to turn on your printer, if you touch the screen while in sleep mode

Without using PSU Control, how can you turn the printer on when touching the screen? Can you set it to use a Custom Action that is a System Command which the wake from sleep uses? My printer is powered on and off via a Tasmota Wifi wall plug, which PSU Control does not support, and the System Commands part of PSU Control is bugged right now. Plus, seems a big pointless to have it installed solely for the purpose of sending a System Command that points to another plugin.