UnchartedBull/OctoDash #698: Proeblems with filemanager / files  are not always displayed

jupzup, 2020-05-31:
I have problems with the filemanager. Sometimes it will be show all files correctly and sometimes i see only the pacman and it it will be only show the message "loading" but no files will be show.

Is this a known behavior ?

greets

OctoPrint v.1.4.0
OctoDash v.1.5.0

UnchartedBull, 2020-05-31:
Had this happen a couple times to me as well. Will do some investigation, but it is most likely a bug within the OctoPrint API.

daimon82, 2020-06-02:
I had the same problem, but somewhat worse. When starting a print form octoprint, prusa only displayed 0% L=-/- and nothing else happened even the heating wouldn't start. 
Both the loading problem and printing was resolved by removing the sd card form prusa.

UnchartedBull, 2020-06-03:
Are you guys having an SD card plugged into the printer? There was a PR merged a couple of weeks back, which added support for SD Cards. One user reported that his problems went away after removing the SD Card from his Prusa MK3 (which of course isn't a permanent solution, but could prove that the SD support actually is at fault here).

Also please send your printers here.

italocjs, 2020-06-06:
I have a similar issue, when i try to click on the "/" (file location, initially at root) the screen freeze, i had been trying to get a file from a pendrive to print, but hadn't been successful, any ideas?

PR #865 (2020-07-31): Add timeout while loading files
If nothing is returned from the API within 6s an error will be shown. While this is not optimal it should work for now. Fixes #698.
closed 2020-07-31 by UnchartedBull