UnchartedBull/OctoDash #699: HyperPixel 4 Flicker/Striping

jneilliii, 2020-05-31:
**What doesn't work?**
When using OctoDash with a HyperPixel 4 there is striping on the display as if there is an overscan issue or something like that. It's like what an old CRT would look like when recorded on video, where a stripe constantly cycles through the screen.

**What did you already try?**
Technically, OctoDash is working, but the striping is a bit annoying. I've tried updating the kernel, the drivers, a few of the config.txt settings.

**General Information:**
 - Hardware Raspberry Pi 4 Model B - 4G RAM, Pimoroni HyperPixel - 4.0", Official Pi4 power supply.
 - OS Info OctoPi 
 - OctoDash Version v1.5.0
 - OctoPrint Version v1.4.0

**Additional context**
I'll try to take a video later but wanted to get this question out there before going to bed. :) I do not see the striping during boot as all the text is scrolling through. I don't have any other desktop installed to try it, if you'd like me to do that I'd be happy to, but would probably need some direction as I've never set that up on an OctoPi setup before.

UnchartedBull, 2020-05-31:
Ok that shouldn't happen. Maybe try unplugging the screen and cleaning all the contacts? Otherwise please send a video for further troubleshooting.

jneilliii, 2020-06-02:
Here's an example video showing the issue. I honestly don't know if this is OctoDash specific or not, but it doesn't seem to be happening on boot up like I mentioned before.

https://youtu.be/3Hg3zKXGpC4

DolfLund, 2020-06-03:
Looks hardware related to me, it doesn’t appear when any other application or terminal is on the screen?

jneilliii, 2020-06-03:
So tonight I completely re-flashed OctoPi on the SD card and after installing the HyperPixel drivers and OctoDash from scratch everything appears to be fine now. I don't know if something got messed up during an OctoDash upgrade from the previous version to the latest or some other random apt upgrade that caused it or what. Hopefully it sticks.

closed 2020-06-03 by jneilliii

UnchartedBull, 2020-06-03:
Let's hope the best, please reopen this issue if it comes back up. I definitely haven't implemented an animation that looks like this, so there maybe was something funky going on with apt update or a slightly corrupted downloaded file.

Jiheffe, 2020-07-04:
Hi @jneilliii , do you have a Hyperpixel 4 touch?
I had a lot of issues getting the Hyperpixel configured correctly. I finallt got it working following Chris' Basement's tutorial for Hyperpixel with Octo**screen**. https://youtu.be/OJ59hXSyBoI . Ak was working fine, including touch, but a sometimes the screen would be shifted to the right.
I did uninstall Octoscreen (sudp dpkg -r Octoscreen) in installed Octodash. It is starting up but I can't get past the initial welcome screebn, because the touch interface is not working...
Will try a clean reinstall on the SD card.

jneilliii, 2020-07-04:
Yep, I have the touch version. The one trick that I think was crucial was installing the experimental touch fix hyperpixel drives for pi 4 and then commenting out something in config.txt. Let me grab that for you.

jneilliii, 2020-07-04:
Here's my config.txt
```
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
#dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
# enable raspicam
start_x=1
gpu_mem=128

dtoverlay=hyperpixel4-common
dtoverlay=hyperpixel4-0x14
dtoverlay=hyperpixel4-0x5d
gpio=0-25=a2
enable_dpi_lcd=1
dpi_group=2
dpi_mode=87
dpi_output_format=0x7f216
dpi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 6
display_rotate=3
```

Jiheffe, 2020-07-04:
I just did aq clean Octopi install on fresh card, installed Hyperpixel with the Experimental touch fix, the changes in config.txt, and then Octoscreen. but the touch interface had not rotated, so I had to add the following line to /usr/share/X11/xorg.conf.d/40-libinput.conf, in the section MatchIsTouchscreen:

Option "TransformationMatrix" "0 -1 1 1 0 0 0 0 1"

JF

jneilliii, 2020-07-04:
I may have also ran the command `DISPLAY=0:0 hyperpixel4-rotate left` during initial setup and that may do something similar to what you did manually. Not really sure what the commands do or if sudo is required or not,