UnchartedBull/OctoDash #700: Cannot run it. Nothing happened after reboot.

vkilin, 2020-05-31:
I ran script from installation tab:
wget -qO- https://github.com/UnchartedBull/OctoDash/raw/master/scripts/install.sh | bash -s -- --ptg
Since I had TouchUI installed I did fully uninstall it by command:
sudo ~/TouchUI-autostart/helpers/uninstall

Now I have raspberry desktop on touchscreen. and if I  ran install Octodash command through Putty it says installed, but nothing changes on touch screen.
Please advice. Octoprint itself works fine through web console.
  

**What did you already try?**
A clear and concise description of what you tried to make OctoDash work.

**General Information:**
 - Hardware [e.g. Raspberry Pi, if you have layout issues please also include your screen resolution]
 - OS Info [e.g. Raspbian Buster Lite, please also indicate if you used OctoPi]
 - OctoDash Version [e.g. v1.0.0]
 - OctoPrint Version [e.g. v1.0.0]

**Additional context**
Add any other context or screenshots about the feature request here.

UnchartedBull, 2020-05-31:
If you want to start OctoDash via putty please execute the following:

```
export DISPLAY=:0
octodash
```

If that works please add octodash to the automatically started applications in your desktop environment configuration (there should be plenty of tutorials for that, maybe search for Raspbian Chrome Kiosk Mode)

vkilin, 2020-05-31:
Yes, I tried this as well. I have this massage:
"(electron) The default value of app.allowRendererProcessReuse is deprecated, it is currently "false".  It will change to be "true" in Electron 9.  For more information please check https://github.com/electron/electron/issues/18397
"

vkilin, 2020-05-31:
It works, since Putty showing some error massage:
(electron) The default value of app.allowRendererProcessReuse is deprecated, it is currently "false".  It will change to be "true" in Electron 9.  For more information please check https://github.com/electron/electron/issues/18397

On the touchscreen I see welcome screen for Octodash! Thanks!
One more question, what is Feed Lenght and Feed speed for Prusa i3 mk3s, maybe you know?

UnchartedBull, 2020-06-01:
Great, that this works for you. Feed Length should be around 30-40mm and Feed Speed should be rather slow (you don't need the filament to travel that much). I use 30mm/s for my Titan Extruder.

closed 2020-06-01 by UnchartedBull