UnchartedBull/OctoDash #701: Add OSOYOO 5 Inch DSI Touch Screen to Supported Devices List

ChaosBlades, 2020-05-31:
I just upgraded to an 'OSOYOO 5 Inch DSI Touch Screen' along with a Raspberry Pi 4 (4GB). This screen works perfectly with OctoDash. A single DSI Cable and no drivers. I installed OctoPi 0.17.0, did a pkg update / upgrade, ran the OctoDash install script, and set the auto login console. That is it, everything works. I also tested with a Pi Camera installed as well and had no issues.

The screen has great colors, text looks sharp, touch is accurate, and the screensaver works (once I remembered it wasn't on by default). Everything feels very snappy and animations are silky smooth.

Capacitive Touch
24-bit color
800x480
60hz

https://www.amazon.com/gp/product/B07KKB5YS9/
https://osoyoo.com/2019/09/20/instruction-for-raspberry-pi-5-dsi-touch-screen/

I would think their 7 inch screen would work the same as well.

tgmorris99, 2020-06-01:
Just installed OctoDash on this display for my BLV mgn Cube build and it works great. All I need now is OctoDash V2.0 so I can dim the too bright background on it.

The only problem is the display cable is straight-thru and the standard cables are twisted. so I cut off one end on the longer cable I needed and stripped off the covering to make it a straight-thru. Works perfectly.

UnchartedBull, 2020-06-01:
Has been added, thanks for the tip!

@tgmorris99 OctoDash v2.0 probably won't have brightness control yet. You can add a line to your `~/.xinitrc` file to automatically set the brightness on each start. This could be useful, if you want to always have your screen on 50% for example. I'll probably use `xbacklight` for OctoDash, so maybe try adding that line to your .xinitrc: https://www.x.org/archive/X11R7.5/doc/man/man1/xbacklight.1.html :)

closed 2020-06-01 by UnchartedBull

tgmorris99, 2020-06-02:
Oh, sorry. by "dim" I meant use the new custom css to change the background color to something darker as I already have my overall brightness turned down.