UnchartedBull/OctoDash #704: Prusa Mk3S 3.90 firmware and touchscreen not switching off and the files screen never lists the files.

skibum-za, 2020-06-01:
I've only recently installed octodash (last week). Anyway, it was working great with my pi4 2gb with my official touchscreen. 

Then I did a few things. Installed a few extra plugin and also updated the printers firmware to 3.90 which was recently release.

So, since doing this the touchscreen always stays on. Even though its configured to turn off :- 

`                       "touchscreen": true,`
`                        "turnScreenOffWhileSleeping": true`

So, the first thing I tried is disabling all the new plugins I've installed and rebooting the pi. But that hasn't help.

As yet I've not tried downgrading the firmware. But I'll probably attempt to do that in a few days assuming I can find the older version.

There is one other thing. When I go to list the files, the little pacman starts eating his dots. But that's it. It never finishes. I've left it there for 10+ mins and still nothing. However, typically a reboot of the pi fixes that for a while. I've got 224 files and they are using 2.8gb.

`OctoPrint version : 1.4.0`
`OctoPi version    : 0.17.0`

UnchartedBull, 2020-06-01:
Did OctoDash enter Sleep Mode? During active mode OctoPrint currently always stays on, if you want to change that behaviour please configure the xorg dpms line in your `~/.xinitrc` file accordingly (or remove it, to use the system defaults).

Downgrading the firmware of your printer probably wouldn't change a thing regarding to OctoDash, so no need for that :).

Regarding the files this issue has been reported in #698 and will be tracked there.

UnchartedBull, 2020-06-22:
File issue will be tracked in linked issue. If you still have the sleep issue please reopen this ticket.

closed 2020-06-22 by UnchartedBull