UnchartedBull/OctoDash #708: Trace/breakpoint trap

alexanderphoenix, 2020-06-02:
Hi all, tried to install OctoDash but could not get it to launch.
Below are the details, any help is appreciated.

**What doesn't work?**
OctoDash isn't starting. Cli throws:
```
Trace/breakpoint trap
```

**What did you already try?**
Installed following the newly updated `install.sh` script:
```
bash <(wget -qO- https://github.com/UnchartedBull/OctoDash/raw/master/scripts/install.sh)
```
and pointed to the virtualenv when asked:
```
? Please specify OctoPrints full virtualenv path manually (no trailing slash) /home/octoprint/OctoPrint/venv
```

**General Information:**
 - Hardware: `Raspberry Pi 3 Model B Rev 1.2`, `HyperPixel 4.0 Touch` (800x480)
 - OS Info: `Raspbian GNU/Linux 9 (stretch)`
 - OctoDash Version: v1.5.0
 - OctoPrint Version: `version 1.3.8`

**Additional context**
If it helps any, this is a line that appears in the `syslog` at each failed run:
```
Jun  2 17:18:56 araneus kernel: [  920.632004] Unhandled prefetch abort: breakpoint debug exception (0x002) at 0x035c4498
```

UnchartedBull, 2020-06-03:
ok that is really weird. I tried the script on my Pi and everything worked fine. Do you see the unhandled prefetch abort during bootup as well? 

Does the old script work for you? You can easily access that through the commit history: https://github.com/UnchartedBull/OctoDash/raw/2f4b88eeb1c78ad60999c16783ce35bc1ee9ef61/scripts/install.sh

alexanderphoenix, 2020-06-04:
That's the first script I tried a couple weeks ago, since my octoprint venv folder seems to be in a non-standard location I had to fix the `$DIRECTORY` in the script first. 
Once that was done, Octodash was throwing the same error as I detailed before, unfortunately.

I can't see any useful information unfortunately in the strace (-f -ff) that I ran:
```
[pid 28814] getpid()                    = 1
[pid 28814] getresuid32([1001], [1001], [1001]) = 0
[pid 28814] getresgid32([1001], [1001], [1001]) = 0
[pid 28814] unshare(CLONE_NEWUSER)      = 0
[pid 28814] access("/proc/self/setgroups", F_OK) = 0
[pid 28814] open("/proc/self/setgroups", O_WRONLY|O_LARGEFILE) = 13
[pid 28814] write(13, "deny", 4)        = 4
[pid 28814] close(13)                   = 0
[pid 28814] open("/proc/self/gid_map", O_WRONLY|O_LARGEFILE) = 13
[pid 28814] write(13, "1001 1001 1\n", 12) = 12
[pid 28814] close(13)                   = 0
[pid 28814] open("/proc/self/uid_map", O_WRONLY|O_LARGEFILE) = 13
[pid 28814] write(13, "1001 1001 1\n", 12) = 12
[pid 28814] close(13)                   = 0
[pid 28814] clone(strace: Process 28817 attached
 <unfinished ...>
[pid 28817] --- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=NULL} ---
[pid 28814] <... clone resumed> <unfinished ...>) = ?
[pid 28811] <... recvmsg resumed> {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="", iov_len=11}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, 0) = 0
[pid 28817] +++ killed by SIGSEGV +++
[pid 28814] +++ killed by SIGSEGV +++
[pid 28811] --- SIGTRAP {si_signo=SIGTRAP, si_code=TRAP_HWBKPT, si_pid=56439960, si_uid=0} ---
[pid 28812] <... poll resumed> <unfinished ...>) = ?
[pid 28812] +++ killed by SIGTRAP +++
+++ killed by SIGTRAP +++
Trace/breakpoint trap
```

UnchartedBull, 2020-06-06:
Ah ok, I thought the script was throwing the error. Maybe try upgrading to Debian Buster? I read that somewhere, that this could be an issue with Electron on outdated distros. Or is there something specific holding you back from upgrading?

alexanderphoenix, 2020-06-07:
> Or is there something specific holding you back from upgrading?
To be honest, part laziness, part if-it-ain't-broke-don't-fix-it mentality.

Good shout though, I upgraded today to Buster following your suggestion.
Unfortunately no joy, it still spits out the same error :-(

alexanderphoenix, 2020-06-07:
Tried reinstalling from scratch (after removing completely) and this is the output:
```
octoprint@araneus:~/git/OctoDash $ bash <(wget -qO- https://github.com/UnchartedBull/OctoDash/raw/master/scripts/install.sh)
Installing OctoDash v1.5.0
Installing Dependencies ...
All packages are up to date.
bc is already the newest version (1.07.1-2).
libappindicator3-1 is already the newest version (0.4.92-7).
libatspi2.0-0 is already the newest version (2.30.0-7).
libnotify4 is already the newest version (0.7.7-4).
libnss3 is already the newest version (2:3.42.1-1+deb10u2).
libsecret-1-0 is already the newest version (0.18.7-1).
libuuid1 is already the newest version (2.33.1-0.1).
libxss1 is already the newest version (1:1.2.3-1).
libxtst6 is already the newest version (2:1.2.3-1).
ratpoison is already the newest version (1.4.9-1).
xinit is already the newest version (1.4.0-1).
xserver-xorg is already the newest version (1:7.7+19+b8).
gir1.2-gnomekeyring-1.0 is already the newest version (3.12.0-1+b1).
libgtk-3-0 is already the newest version (3.24.5-1+rpt2).
x11-xserver-utils is already the newest version (7.7+8+rpi1).
xdg-utils is already the newest version (1.1.3-1+rpt1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Neither /home/pi/OctoPrint/venv nor /home/pi/oprint can be found.
? Please specify OctoPrints full virtualenv path manually (no trailing slash) /home/octoprint/OctoPrint/venv                                                                           /home/octoprint/OctoPrint/venv
? Which plugins should I install (you can also install them via the Octoprint UI)? (Press <space> to select, <enter> to finalize)                                                           Display Layer Progress (mandatory)
Installing Plugins...
Traceback (most recent call last):
  File "/home/octoprint/OctoPrint/venv/bin/pip", line 7, in <module>
    from pip._internal import main
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/pip/_internal/__init__.py", line 5, in <module>
    import logging
  File "/usr/lib/python2.7/logging/__init__.py", line 26, in <module>
    import sys, os, time, cStringIO, traceback, warnings, weakref, collections
  File "/usr/lib/python2.7/weakref.py", line 14, in <module>
    from _weakref import (
ImportError: cannot import name _remove_dead_weakref
Installing OctoDash ...
octodash.deb               100%[=====================================>]  55.35M   494KB/s    in 73s     
Selecting previously unselected package octodash.
(Reading database ... 66203 files and directories currently installed.)
Preparing to unpack octodash.deb ...
Unpacking octodash (1.5.0) ...
Setting up octodash (1.5.0) ...
/var/lib/dpkg/info/octodash.postinst: line 10: update-desktop-database: command not found
Processing triggers for hicolor-icon-theme (0.17-2) ...
Processing triggers for mime-support (3.62) ...
? Should I setup OctoDash to automatically start on boot? yes
yes
Setting up Autostart ...
Setting Permissions ...
OctoDash will start automatically on next reboot. Please ensure that auto-login is enabled!
? Shall I reboot your Pi now? (Use arrow keys)
❯ yes 
  no
```
To note, I only selected the mandatory plugin, after which I got the traceback from the `_remove_dead_weakref` error.

Restarted and tried again, unfortunately I still get the same `Trace/breakpoint trap` error.

UnchartedBull, 2020-06-18:
Hm that python error is related to your OctoPrint installation. Which shouldn't affect the starting of OctoDash at least. I'm really out of ideas by now.

Could you maybe try a fresh install (on another SD Card) and check, whether the error is occurring there as well?

Sorry for my late reply!

alexanderphoenix, 2020-06-20:
Thank you for looking into this.
I might surrender to the idea that there's something weird going on with my octoprint.
I'll give it a fresh install and see if that'll fix the problem!
Cheers

UnchartedBull, 2020-06-20:
Yeah maybe try that on a new SD Card so you can easily roll back :)

Shinjuku1, 2022-05-07:
> Hi all, tried to install OctoDash but could not get it to launch. Below are the details, any help is appreciated.
>
> **What doesn't work?** OctoDash isn't starting. Cli throws:
>
> ```
> Trace/breakpoint [...]

Type in - and hit enter

wrabo, 2022-09-30:
this happen when you change the default user name pi for raspberry pi 
octodash looking for pi as user name when searching for octoprint directory (/home/pi/oprint) you need to change the path with your new user name (replace pi). 
too late i know but maybe someone facing this issue and found this thread :)

BlackBird11, 2022-12-10:
> Type in - and hit enter

this worked for me and as @wrabo explained, yes i change the default username, which is probably the reason why it didn't work