UnchartedBull/OctoDash #710: Turn off display backlight after xx seconds

graphiste, 2020-06-03:
Hello,
My octoprint is aways running and the display stays aways On.
Is it possible to have the display backlight turned Off after some seconds and get it On with touch?
I checked already the available settings, but those dont affect my setup (Octoprint aways running).
I have the official 7" RPi touchscreen V1.1

UnchartedBull, 2020-06-03:
You have to edit your `~/.xinitrc`. Just replace the that reads `xset -dpms` with something like `xset +dpms dpms 600 1200 1800`, just adjust the values to your need - more info available here: https://wiki.archlinux.de/title/DPMS

closed 2020-06-03 by UnchartedBull

RRocks01, 2021-10-02:
> You have to edit your `~/.xinitrc`. Just replace the that reads `xset -dpms` with something like `xset +dpms dpms 600 1200 1800`, just adjust the values to your need - more info available here: https://wiki.archlinux.de/title/DPMS

If its not too much trouble I could really use a dummy's guide for this.  I was able to ssh and set the xset +dpms as above but that will reset on the next reboot.  I don't know how to edit the xinitrc over ssh which is where a dummy's guide would help.  

I really need this screen to go to sleep after a few minutes and then wake up if touched.  I like to do overnight prints but the screen keeps my wife up at night and that's never going to fly, please help, thank you very much.

jneilliii, 2021-10-02:
after SSH to the pi run `sudo nano ~/.xinitrc`, make your changes using your keyboard arrow keys to navigate, press `ctrl +x` to exit, answer `y` to save and press `enter` to overwrite existing file.

7R0J4Ner, 2021-10-03:
> If its not too much trouble I could really use a dummy's guide for this. I was able to ssh and set the xset +dpms as above but that will reset on the next reboot. I don't know how to edit the xinitrc over ssh which is where a dummy's guide would help.
> 

So im a total beginner but just had a similar problem the last view days.

`xset q` just spit out the error message `Error: Can't open display: (null)` for me.
After `export DISPLAY=:0` `xset q` answered something, but this setting was lost at every reboot.
So I wrote `export DISPLAY=:0` in `.bashrc` (with sudo nano ~/.bashrc)
Now you can edit the .xinitrc file with `sudo nano ~/.xinitrc` and replace
`xset -dpms` 
with something like
`xset +dpms`
`xset 60 120 180`

After that you just have to disable the option "Turn screen off while sleeping" in Octodash under Settings or else it will turn off dpms again.