UnchartedBull/OctoDash #713: Can the close button overlay in the internet browser stay while navigating

graphiste, 2020-06-05:
Hi, I use the internate page custom action to open Octprint on the enclosure plugin tab. It works great while navigating between the main Octoprint tabs.
If I click thugh on a link that is not a tab, the new page loads, but the close button desapears and I can nor return to octodash screen.
Is it possible to aways have the close button overlay while navigating internet in kiosk mode?

UnchartedBull, 2020-06-06:
The close button should always stay, so this is a bug, will try to fix it for the next release.

UnchartedBull, 2020-06-22:
The linked PR will fix that issue. The problem is, that there is no way to display a website, if the link wants to open a new window. With the new implementation it will just ignore that link. While this means, that the close button will always be visible, it also means that you probably can't open the website that caused the issue anymore.

Can you maybe update the website to use targer="__self" instead of blank or similar?

graphiste, 2020-06-23:
Thank you very much for solving this issue.

PR #753 (2020-06-23): Fix/no new windows iframe
Fixes #713
closed 2020-06-23 by graphiste