UnchartedBull/OctoDash #716: USB - Pendrive support

italocjs, 2020-06-06:
**Describe the solution you'd like**
A button on the interface with a USB on PENDRIVE "icon", when clicked on it the current location will move to the usb stick and the user will be able to easily print from a usb drive.  Right now i cant print from USB,

**Link other projects**
I'll attach two pictures from astroprint touch, which does support this quite nicely

**Additional context**
It would be really nice to have support to import/print files from a usb stick.  This is specially helpful if there is any king of problem regarding wireless connection (no wifi, not in the same network, so on).  I have been using this feature on astroprint more than sending files over the octoprint site. 
[image]
[image]

I-like-stuff, 2020-06-06:
USB Drive support is not part of octoprint currently so this is not possible unless someone makes a plugin or something like that, here is a link to an issue on octoprint where someone has managed to add it but with some extra configuration needed:
https://github.com/OctoPrint/OctoPrint/issues/1440

UnchartedBull, 2020-06-06:
Until OctoPrint does support pendrives, OctoDash won't support them either. The trick @I-like-stuff posted should work with OctoDash and the files should show up. This will then show a USB directory at root level.

closed 2020-06-06 by UnchartedBull

whoim2, 2020-08-06:
> USB Drive support is not part of octoprint currently so this is not possible unless someone makes a plugin or something like that, here is a link to an issue on octoprint where someone has managed to add it but with some extra configuration needed:
> [OctoPrint/OctoPrint#1440](https://github.com/OctoPrint/OctoPrint/issues/1440)

im set this, but octodash not refresh file list in folder.. OctoPrint see files after refresh button, but octodash - no..

whoim2, 2020-08-06:
> Until OctoPrint does support pendrives, OctoDash won't support them either. The trick @I-like-stuff posted should work with OctoDash and the files should show up. This will then show a USB directory at root level.

please, see comment above

UnchartedBull, 2020-08-10:
OctoDash does refresh the file list each time you change folder (no matter whether it is back or forth). There is a bug in 1.4 that sometimes prevents the API form being updated with the newest files. I think foosel is already working on that. So that hopefully will be fixed in 1.4.1. More info here: https://github.com/OctoPrint/OctoPrint/issues/3422. If the issue still occurs with OctoPrint 1.4.1 I can further investigate this.

whoim2, 2020-08-16:
> OctoDash обновляет список файлов каждый раз, когда вы меняете папку (независимо от того, назад или вперед). В версии 1.4 есть ошибка, которая иногда не позволяет обновлять форму API новейшими файлами. Я думаю, что foosel уже работает над этим. Надеюсь, это будет исправлено в версии 1.4.1. Подробнее здесь: [OctoPrint / OctoPrint # 3422](https://github.com/OctoPrint/OctoPrint/issues/3422) . Если проблема по-прежнему возникает с OctoPrint 1.4.1, я могу ее изучить.

Ok, im update octoprint and see result, tnx!

whoim2, 2020-08-17:
> OctoDash does refresh the file list each time you change folder (no matter whether it is back or forth). There is a bug in 1.4 that sometimes prevents the API form being updated with the newest files. I think foosel is already working on that. So that hopefully will be fixed in 1.4.1. More info here: [OctoPrint/OctoPrint#3422](https://github.com/OctoPrint/OctoPrint/issues/3422). If the issue still occurs with OctoPrint 1.4.1 I can further investigate this.

Im update octoprint to 1.4.2.
Strange, files moving from pendrive mount directory (.octoptint/uploads/USB) to .octoprint/uploads automatically.
I thinks, octoprint move files..