UnchartedBull/OctoDash #717: Cannot open config file via Putty

Pavelovec, 2020-06-06:
Hi,
I'm a beginner in this and I have a problem, that I cannot open octodash config file via Putty with "~/.config/octodash/config.json". It says Permission denied. 

pi@octopi:~ $ ~/.config/octodash/config.json
-bash: /home/pi/.config/octodash/config.json: Permission denied

I want to access config files, because I want to change some custom actions.
Thank you very much for any advice.
PS: I am sorry for my English, it's not my native language.

I-like-stuff, 2020-06-06:
You need to do:
`sudo nano /.config/octodash/config.json`
This will open up the file in nano where you use the arrow keys to move around and Ctrl+X to exit it will then ask you to save if any changes were made: press ‘y’ to save the file or ‘n‘ not to save the file.

UnchartedBull, 2020-06-06:
That is very unusual, you should have full permissions within your home folder. Did you somehow change those (with `chmod` for example)? And yes `sudo` is your friend in this case, thanks @I-like-stuff!

Pavelovec, 2020-06-07:
I tried "sudo nano /.config/octodash/config.json" but it says that "[ Directory '/.config/octodash' does not exist ]".
I think that i didnt change it with chmod (to be honest I do not know what it does), yesterday I updated octodash from version 1.4 to 1.5, before update I could open config as usual, but after update it does not work anymore.
Thank you for any advice

UnchartedBull, 2020-06-07:
Oh I guess there was a typo in i-like-stuffs answer. Needs to be `nano ~/.config/octodash/config.json` and if that doesn't work `sudo nano ~/.config/octodash/config.json`

Pavelovec, 2020-06-07:
> 
> 
> Oh I guess there was a typo in i-like-stuffs answer. Needs to be `nano ~/.config/octodash/config.json` and if that doesn't work `sudo nano ~/.config/octodash/config.json`

This works, thank you very much for everything!

closed 2020-06-07 by Pavelovec