UnchartedBull/OctoDash #720: Display is blank

JeffB42, 2020-06-08:
**What doesn't work?**
I am having an issue like the one reported here:
https://github.com/UnchartedBull/OctoDash/issues/574
(RPi 4 with HyperPixel display, system boots up, rat poison message is displayed, then the screen is blank)

I have found that I can get OctoDash to display if I manually do the following:
$ export DISPLAY=:0
$ octodash

This is not ideal, and I'd like the display to be set and octodash to run automatically.  I looked at .xinitrc and I see that's where octodash runs.  I tried adding "export DISPLAY=:0" to .xinitrc but the screen remained blank.  I also added to rc.local but that didn't seem to have any effect either.  I know this must be possible, but I haven't figured out how (I'm a bit of a Linux noob) and am looking for suggestions on what to try/where to place the command.

UnchartedBull, 2020-06-09:
Have you turned on auto login as explained in the installation guide? https://github.com/UnchartedBull/OctoDash/wiki/Installation#autologin

JeffB42, 2020-06-09:
Thanks!  That was the issue.

closed 2020-06-10 by UnchartedBull