UnchartedBull/OctoDash #725: Install on Windows / Python / OctoPrint 1.4.0?

msteele999, 2020-06-08:
**What doesn't work?**
I am running OctoPrint on Windows 10 Home using virtualenv services. I have the Dashboard plugin (not OctoDash) running and I would like to test your Dashboard.

**What did you already try?**
I am in the discovery phase now - I am hoping I simply 'activate' the virtual environment and run your installer script but wanted to check with you first to see if you have run across this.

**General Information:**
 - Hardware: Intel Compute Stick, Intel Atom x5-Z8330 @1.44GHz
 - OS Info: Windows 10 Home
 - OctoDash Version: Current Version
 - OctoPrint Version: 1.4.0

**Additional context**
Add any other context or screenshots about the feature request here.

UnchartedBull, 2020-06-09:
Won't work OOB no. You'll need to compile the plugin for Windows (currently Linux only). There are tutorials available on electron-builders docs and scripts in the package.json of this repository. Should work without any changes to the code though.

gitxmax, 2020-06-21:
For Atom x5-Z8330 possibly, can install Ubunti, I have done that. Then everything works.

closed 2020-07-22 by stale[bot]