UnchartedBull/OctoDash #726: OctoDash loads but gives different error messages each time I initialise it

stevo1957, 2020-06-09:
**What doesn't work?**
When I load OctoDash I am presented with one of the following messages - 
Can't retrieve layer progress, or
Can't retrieve printer status - 502 Bad Gateway, or
409 conflict

When I initialise OctoDAsh it always says it is asleep and the messages may appear upon wake up.

**What did you already try?**
Reinstalled it - still the same errors.
Basic functionality seems to work though.

**General Information:**
 - Hardware - Raspberry PI 2 Model B
 - OctoDash Version - latestPrusa MK3S printer with MMU2S
 - Octoprint V 1.3.12

stevo1957, 2020-06-10:
Please disregard this issue.
I reinstalled again and things seem to have corrected themselves.

closed 2020-06-10 by stevo1957

AtomicRas, 2021-01-17:
Having this bad gateway 502 issue on a clean install. 

I tried switching the port from 80 to 500 which didn't work just hangs when trying to connect, how to i change the port now if i can't get the Octodash display to load?

UnchartedBull, 2021-01-23:
https://github.com/UnchartedBull/OctoDash/wiki/Setup-&-Settings#manually-adjusting-the-settings