UnchartedBull/OctoDash #738: Gtk WARNING: cannot open display

sblionel, 2020-06-11:
I have a Pi 3B+ and added a [Lebula touchscreen monitor](https://smile.amazon.com/gp/product/B07VNX4ZWY/), connected with HDMI, for use with Octodash. When I try to start octodash I get this:

```
Fontconfig warning: "etc/fonts/fonts.conf", line 100: unknown element "blank"
(octodash:1000): Gtk-WARNING **: cannot open display
```
(The Fontconfig warning can be ignored, as I have read elsewhere.,)

I have gone through the troubleshooting tips (including export DISPLAY=:0), and as many closed issues here as seemed vaguely relevant, but nothing helps. I updated Raspbian, made sure xinit was installed, reinstalled Octodash - nothing changed.

If I run ratpoison, it tells me no display. I am not a Linux expert, but I can stumble my way around. The display works fine as a console. Suggestions welcomed!

sblionel, 2020-06-12:
I got it working... I had to first type `startx`. Then when the X desktop opened, I had to select Run and type `octodash`. How do I get these to run automatically?

UnchartedBull, 2020-06-18:
Hi! Sorry for my late reply.

 Have you enabled automatic login?

You could also try adding that `startx` line to your `~/.xinitrc`.

sblionel, 2020-06-18:
Yes, automatic login is enabled and that works. I would rather not have to do `startx`, as that seems to be overloading the Pi (in that Octodash keeps popping up modal dialogs telling me it can't reach the server, which is the same Pi.) `ratpoison` is supposed to do the job, but it too tells me "cannot open display".

UnchartedBull, 2020-06-19:
What is the error message if you execute:

`export DISPLAY=:0` and `xinit` after that? Could you also make sure, that your display is display 0 and not another number?

sblionel, 2020-06-19:
As best as I can tell, the display is 0.

xinit produces a bunch of messages, ending in:
modprobe: FATAL: Module g2d_23 not found in directory /lib/modules/4.19.66-v7+
xinit: Unable to run program "xterm": No such file or directory
...
xinit: connection to X server lost

A screenshot of this is [here](https://drive.google.com/file/d/1yLZQ5K5dvFSZqKSODBLibyMQDfxSrR0C/view?usp=sharing). 
The Xorg.0.log is [here](https://drive.google.com/file/d/19dt2-960X5s9rIyGEQo24VeSermDp3Ci/view?usp=sharing).

sblionel, 2020-06-19:
Here's something interesting.

I created a file .xinitrc in the pi home folder, put in:
octodash

From an ssh session, I did:
export DISPLAY=:0
xinit

and Octodash starts on the LCD display!

So now I edit .profile to add those commands and reboot. It works!

UnchartedBull, 2020-06-20:
Great! Thanks for the feedback! Closing this here.

closed 2020-06-20 by UnchartedBull