UnchartedBull/OctoDash #745: Display any Sensor Data on Home Screen or Custom Views (i.e. MQTT sub / REST)

Tobi0892, 2020-06-18:
**Is your feature request related to a problem? Please describe.**
As my printer runs in the basement, I constantly monitor it on a tablet in my living room. Hereby, a Python script is running on the OctoPi and collects several printer and print stats as well as data from 7 DHT22 sensors (via GPIO) and sends the data via MQTT to my home assistant system. So far so good.

From these 7 DHT22 sensors, one is installed in the printers' enclosure and 6 in filament dryboxes. The sensors collect the temperature and humidity.

**Describe the solution you'd like**
 Would it be possible that 
- OctoDash receives that data via an MQTT subscription / REST calls or any other method
- and can display any sensor data (name, value, unit) in (a) a separate view or (b) for large i.e. 7 inch displays on the home screen?

**Link other projects**
If interested in directly collecting data from the DHT22 sensors instead of i.e. MQTT as middleware, I can share my repo with the Python code that currently collects the data from the GPIOs via an Adafruit library (was a bit tricky as the pi can not keep up with the timing of the sensors - thus all had to share one ground pin and one gpio that will be pulled low-high to reset all if timing failed. In addition every sensor has a separate data gpio. I.e. if operating 7 sensors one needs 1x ground, 1x gpio for high-low / power and 7x data input).

UnchartedBull, 2020-06-19:
Actually I already thought about that and the configuration file already has 2 filament temperature sensor attributes. So this is / was planned. I'm still not a 100% sure how to do this, but there might be coming something native in the far future!

I'm trying not to include any libraries to read I/O devices from the Pi directly. Had this at first for the temperature sensor and it was a pain, since the packages weren't available for building on Mac and some other inconveniences. Plus OctoDash doesn't use any python code, but relies on TypeScript. The better idea would be to push this as an OctoPrint Plugin, which offers an API / WebSocket that OctoDash can use to display the data, but that is something to think about once I start working on the design for that, which probably will take some time.

As an intermediate solution I propose the following:

- Create a new Lovelace view in HA, housing only the sensor you want to display
- Use the [WEB Custom Action](https://github.com/UnchartedBull/OctoDash/wiki/Custom-Actions#predefined-actions) to load that view
- Log in once (probably need to attach a keyboard and mouse for that)
- Save the login
- Now you should be able to just click an icon in the control section and see all the sensors in your Lovelace View

Let me know if this works :)

gulbrillo, 2020-06-21:
@Tobi0892 I would be HIGHLY interested in your DHT22 Python code! :) I would love to test @UnchartedBull 's proposed web action solution. It sounds perfect as it could go far beyond just temperature and humidity sensors. I could use it to control lights in the enclosure etc.

Tobi0892, 2020-06-21:
@UnchartedBull Thats a good idea. Nevertheless, if I can free up some time, I'll check out the OctoPrint template plugins and investigate how much effort it would be

@gulbrillo 
Here is the code that collects the DHT22, CCS811 (VOC sensor), and print data and sends it via MQTT to Home Assistant:
https://github.com/Tobi0892/octoprint-mqtt-enclosure

Attached the link how to install the adafruit dht library and how the sensor ist connected:
https://tutorials-raspberrypi.de/raspberry-pi-luftfeuchtigkeit-temperatur-messen-dht11-dht22/

gulbrillo, 2020-06-30:
> @gulbrillo
> Here is the code that collects the DHT22, CCS811 (VOC sensor), and print data and sends it via MQTT to Home Assistant:
> https://github.com/Tobi0892/octoprint-mqtt-enclosure

Thanks a lot @Tobi0892 ! :) I am completely new to MQTT / Home Assistant but your script made me look into it. And I love it! I wanted to start with something simpler and decided on an MQTT light switch. I'm also new to Angular, but found a module that allows subscription to MQTT topics. And I hacked it onto the standby screen.
see: https://youtu.be/lrslFFx5XdI :D

@UnchartedBull I managed to get the web action working with Home Assistant in Kiosk mode. That's a really really cool feature!! I'll try messing with the standby screen a bit more to get some native MQTT features on there. Is the github source code already version 2.0? or is this still 1.5?

UnchartedBull, 2020-06-30:
master is always kind of between the releases. So currently most of the 2.0 stuff is there, but a few features are missing. So not really 2.0, but not 1.5 either ...

SaKiEQ, 2021-02-10:
I 'hacked' the enclosure plugin sensor data acquisition function, removed the DHT sensor code and added my external code, once the data formatting was ok it just worked fine. I am getting now enclosure temp via an external source fed into OctoDash.

Not ideal, since it will burn-up once Enclosure plugin gets updated - or I by mistake allow the update happening while not protecting the modification...

UnchartedBull, 2021-02-12:
Well there might be good news for you. With the next release all services communicating with external sources will be put behind an abstract layer, so then you can just pull out Enclosure und plug in your custom integration. Should work fine as long as the abstract service is correctly implemented.