UnchartedBull/OctoDash #746: can't build a working octodash executable

gdachs, 2020-06-19:
**What doesn't work?**
I can't build an OctoDash executable that is working afterwards. It starts with a dialog with an error message that it can't find /src/assets/icon.png. After copying it from the build folder it starts without error message, but only with an empty greyish screen.
   
**What did you already try?**
I use a VirtualBox with Ubuntu, installed nodejs and cloned your git. I used npm install, and npm run start, or npm run pack. The VirtualBox is just fine, it starts OctoDash from you debian package without issues. So there is no problem with runtime dependencies.
  
**General Information:**
 - Hardware just a PC with VirtualBox
 - OS Info I tried Ubuntu 18.04 and 20.04
 - OctoDash Version current master branch
 - OctoPrint Version 1.4.0, but unrelated to the problem

**Additional context**
I suppose I have some missing dependencies, or build steps that I can't figure out. In your .travis.yml you are using nodejs 10, I have tried this too.

UnchartedBull, 2020-06-19:
travis.yml is currently only testing whether something builds. Haven't had the time to improve the pipelines, so it just checks whether everything builds.

`npm run pack` is the right command though. Have you tried installing the .deb file on your ubuntu machine? Maybe also try starting OctoDash via ssh on the Pi to see if the console throws some errors. If all of that doesn't give any information you may also check the developer console in the electron app. One of those should show you the error. Feel free to report back so I can help you with that :)

mikekscholz, 2020-06-19:
I ran a build with the exact same setup as above except with node v12, and got the exact same dialog. I then downloaded the deb from the releases page and installed it over top but the issue persists. I was able to build a functional 1.4.0 on this same virtual machine as an aside.

mikekscholz, 2020-06-20:
so I deleted my node_modules and dist folders, and then switched my node version to 10 via NVM, and then rebuilt, and I no longer get the error dialog, however all I get is the OctoDash icon popping up for about half a second before going to a black screen and then it just stays black. I removed the entry from my xinitrc, updated my pi, and read the install script and saw that you added bc as a dependency since I built this image for octodash testing (like 10 months ago). when I try starting manually over SSH all I get is a segmentation fault.

UnchartedBull, 2020-06-20:
Ok that is all pretty weird. I also run the builds with node 12 without any issues. I‘m using macOS though.

Maybe try purging OctoDash and then reinstalling it from scratch. The release builds should work without any issues.

The bc dependency is for the installer only and not needed by electron afaik. Did you copy the correct architecture (armv7l), If you see the octodash logo that means that the app is starting at least.

mikekscholz, 2020-06-21:
I did that as well, an uninstall, reinstall, then when that didnt work I did a full purge. The only thing I haven't done is wipe the card and start from scratch.

gitxmax, 2020-06-22:
tried compiling for x64 with node12 and got the same error.

UnchartedBull, 2020-06-22:
@mikekscholz so you're still not able to start OctoDash? That is really really weird, since after a purge you should be able to install the official releases just fine.

@gitxmax are you also building this on ubuntu? I might need to investigate this further then, since on macOS there no issue with node v12.16.1.

gitxmax, 2020-06-22:
yep, ubuntu, kernel 5.4 
the only thing I havn't tried is to delete the config folder.

UnchartedBull, 2020-06-22:
Ok, I managed to reproduce this issue. Turns out that Electron 9 changed some stuff, which made loading scripts impossible with the current CSP. I temporarily removed CSP altogether (which might throw a warning during development), but at least everything builds fine for me again and the package can be executed on the Pi. Please reopen if you still have issues once the linked PR is merged (don't forget to pull the latest changes over :)).

You might also want to install `desktop-file-utils`, this throws a warning during installation, I couldn't find a direct link to the failed start issue though.

closed 2020-06-22 by UnchartedBull

mikekscholz, 2020-06-23:
That did the trick, I havent ran a print yet but the build I ran last night appears to be working.

ntrantham, 2020-10-26:
I just did a fresh install of OctoPi and OctoDash and I am getting the segmentation fault when trying to start OctoDash.  Above you mention removing CSP to resolve.  I am not a linux person, is there a step by step procedure for doing this?

UnchartedBull, 2020-10-26:
CSP is still removed from OctoDash at the moment. Is your window manager running and are you exporting the correct display? Also are you starting OctoDash manually or automatically?

ntrantham, 2020-10-27:
I used a default octopi install, installed the display drivers for my 3.5" touch screen.  It is connected via the GPIO pins.  I got the OctoPi booting with the terminal showing on the screen.  Then I ran the OctoDash install.  I have not done anything else.  The install instructions for OctoDash do not mention a window manager.  I don't believe the OctoPi distro comes with a window manager configured.  What do I need to do in order to install?

ntrantham, 2020-10-27:
as far as running manually or automatically.  during the octodash install I selected to have it start automatically.  Now my pi boots, the screen ultimately comes to a cursor in the upper left corner.  If I putty into the octopi and run octodash manually I get the segmentation fault.

mikekscholz, 2020-10-28:
The window manager for octodash is called ratpoison, its super basic, I use it for fullscreen kiosk web browsers too, and it should have been installed by the octodash installer. also I dont beleive its possible to launch octodash over ssh so that error youre getting isnt anything out of the ordinary

mikekscholz, 2020-10-28:
make sure you have a .xinitrc file inside of your ~ directory with the following:

```
#!/bin/sh

xset s off
xset s noblank
xset -dpms

ratpoison&
octodash
```

PR #750 (2020-06-22): Hide sort if no files are available.
Hides the sort button if no files are available. Fixes #733. 

Also disables CSP temporarily, so Electron 9 works again. Closes #746.