UnchartedBull/OctoDash #757: adding terminal interface ?

sacy80, 2020-06-23:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. If applicable also link other issues / bugs here.
Nope.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Adding Terminal interface. so can easily sent command from the tft :)

**Link other projects**
If you want OctoDash to include other OctoPrint plugins or similar please link them here.

**Additional context**
Add any other context or screenshots about the feature request here.

AlexBarbulescu, 2020-08-19:
GCode Viewer would also be nice to have.

Bond-Addict, 2024-07-09:
I'd like to request this feature as well as it's still a relevant need in 2024.