UnchartedBull/OctoDash #759: anyone have install OctoDash on armbian?

sacy80, 2020-06-23:
**What doesn't work?**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
not able to setup OctoDash on armbian

**What did you already try?**
A clear and concise description of what you tried to make OctoDash work.
Clone git and run the ./install.sh

**General Information:**
 - Hardware [e.g. Raspberry Pi, if you have layout issues please also include your screen resolution]
 - OS Info [e.g. Raspbian Buster Lite, please also indicate if you used OctoPi]
 - OctoDash Version [e.g. v1.0.0]
 - OctoPrint Version [e.g. v1.0.0]

OS: Armbian Buster (mainline kernel 5.4.43)
Octoprint: 1.4
SBC: Orangepi PC +

**Additional context**
Add any other context or screenshots about the feature request here.

UnchartedBull, 2020-06-24:
Are you running 64-bit? If so please update the install script to download the package for arm64. If that doesn't fix your problem please provide more logs, it should work fine on Armbian

sacy80, 2020-06-24:
no it 32bit. but guess. i manage to get it install and work. maybe due to the headless armbian didnt have xorg install. after install xorg it work. 

found that this package "gir1.2-gnomekeyring-1.0" not in armbian repo. not able to install. but still work.

Reading package lists... Done
Building dependency tree
Reading state information... Done
Package gir1.2-gnomekeyring-1.0 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'gir1.2-gnomekeyring-1.0' has no installation candidate

jon-daemon, 2020-06-26:
I have an Orange Pi Zero and I'm trying to use a chinese 3.5" touchscreen. I was able to use console view in portrait mode.
Could you give some more info about your setup? Which display did you use?
Did you install OctoDash from source and ignored package gir1.2-gnomekeyring-1.0?
Did you installed anything else besides xorg?
Does the touchscreen work? Did you need any drivers for it?

sacy80, 2020-06-26:
im using octodash + octodash
memory usage 
MiB Mem :    967.9 total,    316.3 free,    281.2 used,    370.3 buff/cache

sbc: orangepi pc plus
armbian: buster_current_5.4.43 (non-deskop version)
octoprint: v1.4
octodash: v1.5
screen: 7inch Hdmi touch LCD

Im using Hdmi touch LCD no driver need. not sure for your 3.5inch touchscreen. 

what i do to get octodash to be install on octoprint:
1. Install **Display Layer Progress (mandatory)**

what i do to get octodash to be install on armbian:
1. install all the dependency -
sudo apt -y install libgtk-3-0 libnotify4 libnss3 libxss1 libxtst6 xdg-utils libatspi2.0-0 libuuid1 libappindicator3-1 libsecret-1-0 xserver-xorg ratpoison x11-xserver-utils xinit libgtk-3-0 xorg

2. sudo nano ~/.xinitrc
- 
#!/bin/sh
xset s off
xset s noblank
xset -dpms
ratpoison&
octodash

3. sudo nano ~/.bashrc
-
if [ -z "\$SSH_CLIENT" ] || [ -z "\$SSH_TTY" ]; then
    xinit -- -nocursor
fi

4. sudo chmod +x ~/.xinitrc

5. sudo chmod ug+s /usr/lib/xorg/Xorg

6. nano /lib/systemd/system/getty@tty1.service.d/20-autologin.conf
-
[Service]
ExecStart=
ExecStart=-/sbin/agetty --autologin pi --noclear %I $TERM

UnchartedBull, 2020-06-29:
thanks for the fix! might be very helpful to other users.

closed 2020-06-29 by UnchartedBull