UnchartedBull/OctoDash #761: Create hook for running a script on wake-up.

MatthewCroughan, 2020-06-24:
There is already a function that tells Octoprint's PSU to power on via the PSUControl plugin for Octoprint, but this is not ideal for my setup, where I want to give people as clean a setup for Octoprint as possible. What I want to do is develop all of my features to run inside of a container where OctoDash is, and launch scripts at different times (hooks) when things occur in OctoDash. 

Octoprint's plugin hierarchy and config files are messy, and I would much prefer handling all these things in the OctoDash container via the network, as otherwise I will never be able to help people update their systems, since it would require overwriting their configs, and there is no way to merge configs cleanly in Octoprint's architecture.

UnchartedBull, 2020-06-29:
Please don't take this personally, but this is a very specific feature for Octobalena and I don't really see another use case for OctoDash were these hooks come in handy. I currently don't have much time to work on OctoDash and I would like to focus my efforts on features that most of the users profit from. Sorry!
So I probably won't work on that. I'm more than happy to accept a (non-breaking) PR for that and merge that into master. Please make sure, that these hooks can be turned off for non balena installs. Another option might be to intercept the docker traffic and use the API of the PSUControl Plugin as the hook. I don't have enough experience with docker to know if this is viable though ...

MatthewCroughan, 2020-06-29:
How does this have anything to do with a platform? 

Is it really that much to ask to have a custom script run when you tap the screen? Come on now. I do take it a bit personally.

UnchartedBull, 2020-06-29:
It isn't related to the platform, but rather to your specific use case. As I stated above this hook will only be beneficial for your project. It isn't that much effort that is true, but it will take some time to do this properly (configuration file, error handling, documentation, ...). There are currently 33 open issues and I struggle to keep up with them. I'm just doing this for fun and I don't get any money from this and offer everything for free here on GitHub, that is one side of the Open Source medal, which I absolutely love and I think more people should Open Source their software so people like you can offer improvements. But Open Source should also encourage people to create PRs for specific features they might profit from. From what I've seen so far this is quite common within the Open Source community, and is also the way I proceeded with the OctoDash development. The DisplayLayerProgress API was also created with a PR from me and another guy who wanted to use it in his Dashboard.

I tried to follow most best practices for Angular & TypeScript and tried to make everything best structured as possible. So please don't take this personally, but I'm currently not able to keep up with all the issues, since I also have other projects / social life. This whole project originally was designed as a weekend thing and it kind of exploded, which is awesome! But there just is lots of stuff going on and I try to focus on the most requested features, but I also need to stuff that I enjoy. If I don't enjoy working on this project anymore, I'll probably drop it and I think neither of us want that.

If you'd like I can reopen this issue and attach the help-wanted label, so maybe a community member can chime in here. I just don't have the capacity to work on this. V2 should be released for 6 months or so by now and yeah, we're still at 1.x. Please don't take this personally, I just have to find a way to keep up with the growing number of feature requests.

Again, sorry if this has upset you, but this is also a great way to get to learn node. I'm more than happy to help out if you hit a snag and review your PR, I just don't have the time to do this on my own and I don't want to create false hopes and then just ignore this issue. Hope you understand.

Cheers

closed 2020-07-29 by stale[bot]