UnchartedBull/OctoDash #762: Use onscreen keyboard or other input method to define config file

MatthewCroughan, 2020-06-24:
It would be great if the config.json file were definable entirely by touching the screen, or a web interface so that noobs could easily configure this forever, without ever having to interface with the CLI.

MatthewCroughan, 2020-06-24:
https://www.npmjs.com/package/electron-virtual-keyboard

UnchartedBull, 2020-06-29:
Good idea. I like the design of this one a little better though: https://www.npmjs.com/package/simple-keyboard. Leaving both here for now.

dintxaustip, 2020-08-02:
Hola a todos. Soy un absoluto novato, pero después de 20 intentos (y no exagero) he conseguido instalar octodash en mi pantalla MHS-3.5. Lo primero que conseguí fue ver la interfaz de raspbian en mi lcd. Cambié en Raspi-config para que se cargara automáticamente al arrancar y actualmente me muestra el saludo de Octodash para configurarlo. Problema: no tengo teclado en pantalla para completar la configuración. Entiendo que se puede instalar un teclado virtual, pero no sé cómo hacerlo. Me gustaría que apareciera en pantalla cuando pretenda rellenar los huecos de texto. Utilizo Putty para meter toda la información. Qué órdenes debería copiar?. Agradezco de antemano su atención y espero me puedan ayudar. Gracias

UnchartedBull, 2020-08-05:
I can't exactly can help you with the on-screen keyboard, but you can also use xdotools as outlined here: https://github.com/UnchartedBull/OctoDash/wiki/Setup-&-Settings#setup-without-keyboard. There are also multiple tutorials available on how to use xdotools around the internet :)

Also temporarily attaching a keyboard (only needed for setup) is an option as well :)

TheSin-, 2020-08-11:
Just as an idea what is octodash had an octoprint plugin, so you could enter everything (except the api info) via the plugin (web ui), you could also use the plugin to add any info you want form octoprint, printer name, extruder count and setup, etc etc

Just an idea, I was working on adding the virtual KB but I don' think it's going to work since the display is so small it's a difficult task, this might make the best of both worlds.  As for the API info you could read it from a special file on /boot so when you setup octopi and you need to mode the spa info text file we could have an octodash txt file there as well that get copied into place on boot.

Again these are just ideas to try to make this great project easy to setup.

UnchartedBull, 2020-08-18:
I also thought about that plugin idea. I might need to checkout whether it is somehow possible to bootstrap the installation, so that if you install the plugin OctoDash gets installed as well. Needs to be properly documented and stuff of course, but this is definitely something for one of the next versions.

Regarding the virtual Keyboard, I'm planning to integrate one natively, but on a 3.5" this is most likely too small to be useful. So the plugin might be a good fit here.

TheSin-, 2020-08-18:
yeah after all my tries with different on screen KBs even in the large mode it's not practical at all, I think a plugin is more and more the way to go, I'm almost done the design on my new printer and I plan to start on such a plugin and add a PR for it.  I'm hoping to deal with multi hotends as well since I plan to get the printer config from Octoprint.  I do not want to fork the project though you have done great things here I'd just like to contribute and help out where I can with it.

So if you don't like my work that isn't a problem at all at least I hope to give you a base to work with cause this is an important project IMHO.

UnchartedBull, 2020-08-19:
Printer Config already is implemented (rudimentary) to get axis inversion, so you could build on that I guess. And yeah as I mentioned OctoPrint plugin is on the list, I still want to try the simple-keyboard and at least check if it's a viable options for 5"/7" display. Or maybe for only entering numbers.

What do you mean by you don't want to fork the project? Would be more than happy to review/merge a PR for a plugin / multi extrusion! :)

TheSin-, 2020-08-19:
Yeah I'll make PRs I just meant I didn't want to fork it and make my own changes to turn it into an other project I'd rather help here.

If we can get all he print info that means you can get the active printer profile and name, so when you switch it in octoprint it'll switch in octodash which is great, so only the api url and key should be required, should be a lot easier to enter info to set it up since it's just 2 fields.  Then everything else is from OctoPrint already or anything else we need via the plugin and it's easier for everyone to use and setup out of the box which I think would be perfect.

Since you have a plan for the KB I think I'll start looking more at the config and what we can get from octoPrint to avoid extra input to start.  I haven't really looked deep in the api yet but I was playing around with the printer config list and seeing which one is the currently active profile, which does contain the the extruder count.

Since I currently use an MKS TFT I can use some of the UI elements from it.  For example to see the temps and load each one you click the extruder icon and there is a small number that changes, kinda like cycling through them.  That way the UI doesn't skew or get too squished.  That being said I feel I might be high jacking the Original topic for this ticket so maybe I'll start a new one for the OctoPrint Plugin and Config rework for less inputs.

UnchartedBull, 2020-08-19:
I would be more than happy to accept your PRs :) Once I've seen a few PRs from you I can also add you as a contributer to the project if you like, then you wouldn't to compare across forks all the time (and this would really help me out).

Regarding the API there are some changes planned (especially switching over to the push data model, I'll post some more info over in the new issue).

I really like the idea with the rotating extruders, since that wouldn't break everything for a single extruder setup and would keep the clean look.

7R0J4Ner, 2021-09-03:
Hello all.
I just stumbled here by chance via Google.

For all those who possibly also happens:
[OctoDash Companion](https://github.com/jneilliii/OctoPrint-OctoDashCompanion)