UnchartedBull/OctoDash #767: Wifi Manager at start for Touchscreens

Kefier, 2020-06-26:
It would be nice if OctoDash has a Wifi Setup Menu at the startup so you can login into your Wlan. Like in Octoscreen, where you can easy scan the wifi area and connect. This and an implemented keyboard to write on a touchscreen would make octodash perfect. :*

UnchartedBull, 2020-06-29:
Keyboard is tracked in #762 and definitely will come. Regarding the WiFi thing - I have to check whether that is possible with electron and how much effort that will be. If it is possible, it shouldn't be a big problem to implement though.

charlespick, 2021-05-05:
I haven't been able to find a way to get electron to work with wifi directly. How do we feel about using a Node backend to perform more system stuff in Octodash? It could be optional, control wifi, apt updates, etc. Wifi would be easy to implement using [node-wifi](https://github.com/friedrith/node-wifi).

UnchartedBull, 2021-05-05:
Yes `node-wifi` is definitely the way to go. Somebody mentioned ~2 weeks ago, that they'll take care of that. I haven't heard anything from them yet though. We decided on `node-wifi` though :)