UnchartedBull/OctoDash #773: Type 'Timeout' is not assignable to type 'number'.

gulbrillo, 2020-06-30:
**Describe the bug**
Source code produces a type error.

**To Reproduce**
`npm run start` results in:
```
    ERROR in src/app/filament/filament.component.ts:202:9 - error TS2322: Type 'Timeout' is not assignable to type 'number'.
    202         this.timeout = setTimeout((): void => {
    src/app/filament/filament.component.ts:221:9 - error TS2322: Type 'Timeout' is not assignable to type 'number'.
    221         this.timeout = setTimeout((): void => {
    src/app/filament/filament.component.ts:224:11 - error TS2322: Type 'Timeout' is not assignable to type 'number'.
    224           this.timeout2 = setTimeout((): void => {
    src/app/filament/filament.component.ts:292:7 - error TS2322: Type 'Timeout' is not assignable to type 'number'.
    292       this.timeout = setTimeout(this.automaticHeatingStartTimer.bind(this), 1000);
    src/app/filament/filament.component.ts:306:7 - error TS2322: Type 'Timeout' is not assignable to type 'number'.
    306       this.timeout2 = setTimeout(this.checkTemperature.bind(this), 1500);
    src/app/filament/filament.component.ts:314:7 - error TS2322: Type 'Timeout' is not assignable to type 'number'.
    314       this.timeout2 = setTimeout(this.checkTemperature.bind(this), 1500);
```

**Fix**
Replace `setTimeout` with `window.setTimeout` for the six instances mentioned above.

```javascript
  public ngOnInit(): void {
    if (this.configService.getAutomaticScreenSleep()) {
      this.displaySleepTimeout = window.setTimeout(this.service.turnDisplayOff.bind(this.service), 300000);
    }
  }
```

UnchartedBull, 2020-06-30:
Does that error occur only after you installed the MQTT package? Most likely that MQTT package has some node dependencies, which caused the mixup with the TS timer as explained here: https://github.com/microsoft/TypeScript/issues/32133.

gulbrillo, 2020-06-30:
Interesting, yes! It started to pop up along the way after I installed a few extra packages (I added the material grid view, ngx-mqtt, and rxjs.) The `window.` fixed it for me. It could be that one of the packages added a node-side `setTimeout()` definition. (?)
I'm used to Vuejs. All of this Angular stuff is super new and very confusing to me. Adding functionality without breaking other stuff seems to be harder than expected. :/

UnchartedBull, 2020-06-30:
Yeah the „problem“ is that the MQTT package probably does have a node dependency. Node.js and TypeScript each define their own setTimeout function (one returning a TimeOut and the other one returning a number). If the node types are included (due to MQTT) the node setTimeout will be used by default, thus the error you‘re getting. With window.setTimeout you‘re explicitly calling the TS implementation. Node and TypeScript can be very confusing haha. Closing this here for now, if you want to create a PR please do add the window. :). Thanks! Or do you need the window. to be in master regardless of a possible PR?

closed 2020-06-30 by UnchartedBull

gulbrillo, 2020-07-01:
Took me like 15 minutes to figure out you were not talking about a press release. :D
I'll include all the changes in the pull request. (Although I was hoping to touch as little files as possible.)