UnchartedBull/OctoDash #779: Dual Extruder

schnudergof, 2020-07-02:
Hey there

I'd love to see support for dual-extruder (Show/set Temp, load filament).

UnchartedBull, 2020-07-05:
Would love to work on this, but as stated in #332 I don't have a multi-extruder printer to test this on, which makes it really hard for me to work on this for now.

mkuf, 2020-09-18:
> Would love to work on this, but as stated in #332 I don't have a multi-extruder printer to test this on, which makes it really hard for me to work on this for now.

Octoprint's Virtual Printer Plugin might be helpful here. That Printer can be configured to have multiple Tools. 
https://docs.octoprint.org/en/master/development/virtual_printer.html

-mkuf

radaiko, 2020-12-31:
+1 

I have to use OctoScreen because of Dual Extruder support.

mattydboom, 2021-05-23:
Shame, I (and many others) have dual nozzle printers, and being able to load, unload, and preheat specific nozzles would be a great addition. Before researching this post I thought it was bound to already exist, but seems not. Gets a vote from me!

Nathan22211, 2022-04-09:
Any update on this? I've got a Zonestar with 4 extruders and 1 hotend so I kinda want this to be a thing. Geeetech printers should be cheap, though not great from what I hear, but good enough for testing.

queengooborg, 2022-08-03:
I'm currently working on a pull request to incorporate this functionality for multiple extruder/hotend support (AKA not just two), hoping to have it ready for review and merging soon for all of the fellow multi-extruder printer users out there!

Sneak peek of my progress thus far:
[image]

morphias2004, 2024-10-19:
@queengooborg how are you progressing with this?

I'm looking for a solution for my Prusa XL. I worked with another user and Gina from OctoPrint to have full multi tool support incorporated into OctoPrint.

https://github.com/OctoPrint/OctoPrint/issues/5018

queengooborg, 2024-10-19:
Hi there! I am actually no longer maintaining this pull request, as I have switched to using a Bambu Lab X1 Carbon, and no longer need to use OctoPrint. Additionally, it appears that OctoPrint is unmaintained as there is no new releases since three years ago, so I do not think this pull request (or anything similar) will get merged unless someone forks the project.

PR #3256 (2025-03-17): Add multi-extruder support
This PR resolves #779 (and fixes #4272, a duplicate) by adding multi-extruder support.  This adds support for viewing and setting the temperatures of hotends besides tool 0, adds tool selection to the filament changer and control screens, and displays which filament is loaded into which tool.

The layout has been tested and optimized for one through five toolheads.

<details>
<summary><b>Click to show screenshots</b></summary>
[image]
[image]
[image]
[image]
[image]
[image]
<img width="1140" alt="Screenshot 2025-03-13 at 19 27 21" src="https://github.com/user-attachm [... cut]