UnchartedBull/OctoDash #781: Raspberry Pi 4B Compatability?

kats-greg, 2020-07-03:
Does OctoDash run on a Raspberry Pi 4B?  I have been unable to get it to display on my new pi 4B and new touchscreen.

Are there log files somewhere to look at for errors?  I have tried all the troubleshooting options on the wiki page.

Thanks

OPatrick, 2020-07-04:
Had it working twice with a Pi 4B, with HDMI 7", but since I switched power off without properly quitting, I cannot any more launch it...

UnchartedBull, 2020-07-05:
Are you running the 64bit version? If so you need to download the arm64 package and install that over the current installation (`sudo dpkg -i <downloaded-file`). If you're using the right architecture you might try executing the command `octodash` via the CLI, this should yield some errors then. Just make sure to select the correct display first (`export DISPLAY=:0` should work in most cases)

There also is an issue with some video drivers where OctoDash gets displayed on the HDMI port only. So that maybe also worth a try checking.

@OPatrick that shouldn't happen ... I turn off the power every time without quitting anything and haven't had a problem. Could you maybe check your `~/.xinitrc` to see if the `octodash` command got removed somehow? Might also be worth a shot to execute `octodash` manually as stated above.

OPatrick, 2020-07-06:
> @OPatrick that shouldn't happen ... I turn off the power every time without quitting anything and haven't had a problem. Could you maybe check your `~/.xinitrc` to see if the `octodash` command got removed somehow? Might also be worth a shot to execute `octodash` manually as stated above.

Thanks for your answer. The contend of ~/.xinitrc is: 
#!/bin/sh
xset s off
xset s noblank
xset -dpms
export DISPLAY=:0
ratpoison&
octodash

When executing octodash by ssh, I got:
(electron) The default value of app.allowRendererProcessReuse is deprecated, it is currently "false".  It will change to be "true" in Electron 9.  For more information please check https://github.com/electron/electron/issues/18397
libGL error: failed to create dri screen
libGL error: failed to load driver: vc4
libGL error: failed to create dri screen
libGL error: failed to load driver: vc4
Thanks for your help.

OPatrick, 2020-07-06:
As a side remark, install script modified twice ~/.bashrc (since I re-installed):
if [ -z "\$SSH_CLIENT" ] || [ -z "\$SSH_TTY" ]; then
    xinit -- -nocursor
fi
if [ -z "\$SSH_CLIENT" ] || [ -z "\$SSH_TTY" ]; then
    xinit -- -nocursor
fi
But this is not the cause of my problem because even when I removed redundancy, same black screen...

UnchartedBull, 2020-07-06:
The second part of the bashrc can be safely removed as it is redundant.

The errors you're getting are indicating that there is something wrong with your video driver. Maybe try reinstalling vc4 like mentioned here would help: https://wiki.gentoo.org/wiki/Raspberry_Pi_VC4. They also mention an option in the raspi-config so maybe just reenable the driver via that and see if that fixes the problem.

OPatrick, 2020-07-06:
Thanks, I will try that tomorrow morning.

OPatrick, 2020-07-07:
OK, works fine with Pi 4B 2Gb with Touch screen 7" via HDMI+USB.
To do so, in /boot/config.txt, we uncomment the first line "hdmi_safe=1".
That's the only thing I did.
I would like to thanks UnchartedBull for the outstanding work and the platinium class support.
Octodash is clean and smooth.
Because it is made with Angular, I might go and customise to add enclosure commands like light, heating, air purification, right from the first screen (to reflect the octoprint "enclosure" plugin).
Thanks thanks, you rock!

UnchartedBull, 2020-07-08:
Good to hear that everything is working for you now! Feel free to create a PR for your changes, I'll be happy to merge them into master :)

@kats-greg did you fix your problem? If so please close this issue :)

closed 2020-07-08 by kats-greg