UnchartedBull/OctoDash #783: Screen Calibration

Hawkins00000, 2020-07-04:
Is there an easy way to temporarily disable OctoDash from launching, so that I can run the calibration utility, then re-enable it, so it auto boots into the tool? 

I have a 5in HDMI screen that is working with a few HTTP errors, that sporadically appear,... but my current issue is touch panel caligration.

Sorry< I am a noob, and am not familiar enough with now your awesome tool is integrated.

Great tool though! TY

UnchartedBull, 2020-07-05:
You could either edit your `~/.xinitrc` and put a `#` in front of the line that reads octodash (which will prevent OctoDash from starting up). Or you could get the process id (i.e. with `pgrep octodash`) and then just use `kill <process id>` to stop OctoDash while running.

UnchartedBull, 2020-07-05:
There is also a custom action available to stop OctoDash: https://github.com/UnchartedBull/OctoDash/wiki/Custom-Actions#predefined-actions (`[!STOPDASHBOARD]`)

closed 2020-07-10 by UnchartedBull