UnchartedBull/OctoDash #784: Touch interface not working on Hyperpixel 4

Jiheffe, 2020-07-04:
Hi,
Trying to get Octodash up and running of a Pi 4B witg a Pimoroni Hyperpixel 4 Touch screen.
I first followed the excellent guide over at Chris' Basement (https://youtu.be/OJ59hXSyBoI) to get the hardware set up correctly with Octo**screen**. Worked perfectly except for the screen being randomly shifted to the right. Went ahead and remove octoscreen with sudo dpkg -r octoscreen, then installed Octodash using the automatic script, and enabled console autologin.
At boot, Octodash starts, but the touch interface is not working (was working with Octoscreen). 

Raspberry Pi 4B / 4 Gb
Octoprint 1.4.0
Octopi 0.17.0
Octodash 1.5.O

Something else I need to configure?

closed 2020-07-04 by Jiheffe

Jiheffe, 2020-07-04:
Found an answer in another issue (699). Sorry for this.

UnchartedBull, 2020-07-05:
Glad you figured it out. No need to be sorry though, the setup can be kind of wonky sometimes (since there are just soooooo many screens out there) and this is the best place to ask :)