UnchartedBull/OctoDash #786: Noobie

clue3030, 2020-07-06:
I am a complete noobie when it comes to a raspberry pi. How do I SSH into my rpi4? I have not a clue. I have octoprint already installed and setup for my WiFi. That’s all that’s been done and I did install some plug-ins but that was just click install, and it did it for me. Now I want to use my 7” touchscreen with it. It’s the official touchscreen. But I don’t Know shit about SSH -ing into my rpi4. Please help and give me some instructions. Thank you

amercer86, 2020-07-06:
[Here's](https://www.youtube.com/watch?v=h3_zNvAvJtE) a video that covers the basics of Putty. You'll need to know your Pi's IP and SSH needs to be enabled on your Pi.

As an aside, you don't actually NEED to use SSH if you have your Pi attached to a screen and keyboard. Really, all you need is access to the terminal/command prompt. SSH just allows you to do that over the network.

clue3030, 2020-07-06:
Terminal command prompt on what, the rpi4 or on my Mac computer?

Dryphter, 2020-07-07:
Mac. But you need to enable SSH on the pi. Google is your friend as there are plenty of how-to articles and videos on setting up a pi with SSH.

UnchartedBull, 2020-07-08:
I guess the official docs are a good starting points: https://www.raspberrypi.org/documentation/remote-access/ssh/. There are also lots of videos available online

closed 2020-07-08 by UnchartedBull