UnchartedBull/OctoDash #796: Update to Version 2?

nicedevil007, 2020-07-08:
Hey guys, maybe I have tomatoes on my eyes. Found a update line with the 2.0 version in the wiki here and this page isn't available anymore. Then I read that the new version should be release in a week or 2 (that should me start of june this year?).

I guess something went wrong and it wasn't released yet?

Is there a guide out how to compile it by myself?

UnchartedBull, 2020-07-08:
Hi,

sorry for the confusion. Another (more important project) popped up and i needed to invest my time into that project. Since the other project is almost done by now I'm slowly picking up OctoDash again. I also decided to add more features before the v2 release (on-screen keyboard, update from within the app) - while they make OctoDash more user friendly they take a good chunk of time to get done. So it will be another 2 weeks or month until v2 release, I don't really know ...

If you want to build the latest available version you can always run:
- `npm install`
- `npm run pack`

This should create a folder package with a .deb file in there which you can install normally on your Raspberry.

nicedevil007, 2020-07-09:
Hey @UnchartedBull,

thank you for your respone. Even this answer let me hope :)
I recently switched from OctoScreen to your implementation! OctoScreen always keeps my OctoPrint stopping after I power off the printer :(

Like it a lot, just few things I have to understand.

For this issue, maybe anyone else can help as well:

I have octopi 0.18 installed.

I did the following things:

```
sudo su -
apt update && sudo apt install npm -y
git clone https://github.com/UnchartedBull/OctoDash.git
cd OctoDash/
npm install
npm run pack
```

Then this error appears after some time:

```
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! octodash@1.5.0 pack: `npm run ng:build && electron-builder build -l`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the octodash@1.5.0 pack script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-07-09T12_52_10_056Z-debug.log
```

UnchartedBull, 2020-07-10:
Did you build this on the Raspberry Pi? The Pi might not have enough RAM to build & package the whole application (at least the 1GB models). Please try building on your PC in this case :)

Also the whole log would be really helpful, since the actual error is above the lines you posted.

nicedevil007, 2020-07-12:
@UnchartedBull heya again, I'm using Raspberry Pi 4 with 4 GB, mhhh then I need to try it again on my server with 16 gigs of ram... or maybe I just wait until you release newer version :P

I would close this issue here and post my other existing ones now seperate

closed 2020-07-12 by nicedevil007