UnchartedBull/OctoDash #807: OctoDash does not sleep anymore

mithkr, 2020-07-10:
**Describe the bug**
After update to v1.5.0, OctoDash does not sleep anymore when idle for a while.

**Expected behavior**
After some idle time, the previous version entered sleep mode and showed special sleep screen.
 
**General Information:**
 - Hardware: Raspberry Pi 3 A+  and 5' TFT DSI touch display from DFRobot
 - OS Info: OctoPi 0.17.0 / debian buster 10.4
 - OctoDash Version: v1.5.0
 - OctoPrint Version: v1.4.0

ChaosBlades, 2020-07-11:
I don't have any issues. I think backing up your config and doing a quick setup from scratch would resolve your issue. My 2nd guess is a miss configuration or issue with PSU Control.

UnchartedBull, 2020-07-13:
Please double check whether your screen supports DPMS (if `xset dpms force standby` blanks your screen everything should work). OctoDash will only turn off the screen in sleep mode. sleep mode currently only is activated, once the printer is disconnected from OctoPrint. If the printer is on, OctoDash will also be always on.

ChaosBlades, 2020-07-13:
That's interesting because mine will go to sleep while printing... Sometimes.

UnchartedBull, 2020-07-13:
That actually is a bug that will be fixed in the next version. If you enter and leave standby mode within 5 minutes dpms won't be deactivated, which will lead to dpms using the default settings. If you want to keep that behaviour you can always adjust your `~/.xinitrc` file and define the times in there. Just make sure to disable screen sleep within OctoDash then, since that will overwrite your .xinitrc settings

mithkr, 2020-07-15:
OK, I understood now, that OctoDash is only going to sleep mode if printer is powered off.
This works for me, but would it be possible to add feature to OctoDash, that sleep mode is also activated if printer is power on but idle (not printing) and there was not manual interaction on the touch screen for a while (e.g. 5 minutes or configurable timeout).

UnchartedBull, 2020-07-15:
This will come in the future. Will be tracked in #410 (since that will enable me to do exactly that). Closing this here now, if you need more help let me know :)

closed 2020-07-15 by UnchartedBull