UnchartedBull/OctoDash #808: Hyperpixel 4.0 square?

psk-sudo, 2020-07-10:
Amazing job on OctoDash. Im putting together a part list with the idea of using this in the near future with a RPi4 and had a question about touch displays...

Cant seem to fine anyone who is using the Hyperpixel 4.0 square screen (720x720). Curious if this works as is or if it would need some tweaking outside of the documented resolution adjustments in the terminal? For my printer setup, I'd really like to go with a square display for the aesthetics of course :)

Thanks!

UnchartedBull, 2020-07-13:
Should work fairly fine. Another guy did the same thing and he mentioned that only a few minor adjustments are needed (#579). Square Aspect Ratio screens aren't officially supported though. Maybe you can try talking to the author of the other issue.

closed 2020-07-13 by UnchartedBull

psk-sudo, 2020-07-13:
Thanks! Ill see if I can get in touch.

UnchartedBull, 2020-07-13:
Let me know if you need further help :)