UnchartedBull/OctoDash #809: Screensaver

niklask52t, 2020-07-11:
Is it possible that the screen turns off after a while? and also in the print monitor?

ChaosBlades, 2020-07-11:
Both my adafruit pitft plus 3.5" and OSOYOO 5 inch DSI Touch Screen turn off the backlight after a while. My best guess is that whatever screen you are using isn't supported and you will need to get a different screen.

Same goes for the screens in the Supported Devices Wiki page.
https://github.com/UnchartedBull/OctoDash/wiki/Supported-Devices

UnchartedBull, 2020-07-13:
There are 2 options:

- enable screen sleep in OctoDash (only turns off the screen if disconnected and sleep mode is active)
- adjust your `~/.xinitrc` and set up DPMS normally (disable OctoDash's sleep feature then) - this will then always work and turn off the screen after the set amount of time.

closed 2020-08-12 by stale[bot]