UnchartedBull/OctoDash #810: Tp-link Plugin

hayder9445, 2020-07-11:
Hi,

Is there anyway to combine the tp link plugin with Octodash? With the custom buttons feature is it possible to set up a button to turn on the printer via tp link?

UnchartedBull, 2020-07-13:
OctoDash currently only supports the PSUPlugin (which offers a custom action), but the M80 will work just fine as well.

For controlling the extruder and stepper motors you would need to setup custom buttons yes.

hayder9445, 2020-07-13:
Thanks, I managed to create custom buttons for extruder and stepper motors. 

The problem with M80 is that when the printer is off, octodash doesn't display the UI, only a message saying it can't connect to the printer ( I disabled usb power ).

AdmiralMichael, 2020-07-20:
I too would like to see an option for tp-link smart plug control. It would be great to be able to turn the smart plug off or on, even when octodash is in sleep mode.

UnchartedBull, 2020-07-20:
If you switch over to PSUControl Plugin that is already supported, also when the printer is off. There is an option available, which will tell the PSUControl to turn on the printer before trying to connect to it. The TP-Link plugin currently isn't supported though. As a workaround you can just let the PSUControl execute the M80 command. Haven't tested that, but it should work: https://github.com/kantlivelong/OctoPrint-PSUControl/wiki/Settings

nowkillkennys, 2022-05-21:
how do I make the uplink work with octodash