UnchartedBull/OctoDash #812: Ultimaker Format Package OctoPrint Plugin is required to support UFP files in OctoPrint

ChaosBlades, 2020-07-11:
In order to utilize OctoDash's ability to use ufp files the user would need to install the Ultimaker Format Package OctoPrint Plugin. Since other plugins are automatically installed on installation this one might as well be as well. Otherwise, OctoPrint shows and error when you try to upload upf files.

https://github.com/jneilliii/OctoPrint-UltimakerFormatPackage

UnchartedBull, 2020-07-13:
The new installation script offers the option to install the UFP plugin. Not everyone uses Cura though, so there is no need to make this mandatory, as it is useless for other slicers. Also the plugins page in the wiki offers some info there. That should be enough, shouldn't it?

ChaosBlades, 2020-07-13:
Last time I installed must have been before the new script.. Thats what I was asking for.

closed 2020-07-13 by ChaosBlades