UnchartedBull/OctoDash #813: screen not displaying

wolverine969, 2020-07-11:
Hi,

I managed to install OctoDash and the able to view the setup on my tv. I can't seem to get my kedei 3.5 screen to show octodash. Is there something that I'm forgetting to install? I already installed the necessary display drivers

UnchartedBull, 2020-07-13:
Does that mean you can see OctoDash on an HDMI monitor, but not on the other screen? Also check the troubleshooting guide please: https://github.com/UnchartedBull/OctoDash/wiki/Troubleshooting#cannot-open-display-xinit-failing-

TheLion, 2020-07-25:
I have, sort of, the same issue, using the same display.

Octodash won't auto-start. When running octodash manually, the screen loads on the screen without issues.

UnchartedBull, 2020-07-26:
Could you confirm that your `~./xinitrc` does contain `octodash` somewhere? If not please add that command and try again.

TheLion, 2020-07-26:
> 
> 
> Could you confirm that your `~./xinitrc` does contain `octodash` somewhere? If not please add that command and try again.

Yes, absolutely. If I run `~./xinitrc` manually, it starts just fine (if I put "export DISPLAY=:0" in `~./xinitrc` too).

TheLion, 2020-07-26:
I do seem to have the same issue when I disconnect the Kedei screen and connect a normal HDMI monitor. The Desktop appears and works, but no Octodash.
When I run Octodash from the "Start Menu" it also starts without issues

Seems like `~./xinitrc` doesn't run, or something.

UnchartedBull, 2020-07-28:
Oh wait do you have the normal Raspberry Dekstop installed? If so please enable auto start as described here: https://bigl.es/tooling-tuesday-auto-start-a-gui-application-in-raspbian/ or any other tutorial on the web that uses the raspbian desktop.

TheLion, 2020-07-28:
Yeah! That fixed it.

Don't know if this fixes it for @wolverine969 as well.

wolverine969, 2020-07-28:
Hi,

I’m still having an issue. Is it because I installed octopi?

Regards
Viresh

UnchartedBull, 2020-07-28:
Octopi shouldn't be an issue, there are quite a few people running OctoDash on OctoPi just fine. Did you change any settings during the driver installation with `raspi-config`? If so please revert them, there were a few issues with the exact same problem that was caused by the instructions some sellers did send them. Btw you did follow the troubleshooting guide, right? https://github.com/UnchartedBull/OctoDash/wiki/Troubleshooting#cannot-open-display-xinit-failing-

wolverine969, 2020-07-28:
The problem is that nothing is displaying on the scree . Not sure what drivers I need to install in order to get the display to work

TxBillbr, 2020-08-05:
Not sure what screen you are trying to install. Raspian upgrade to 5.4 kernel is causing problems for some i2c and spi devices. I have been working with Pimoroni folks in getting their HyperPixel 4.0 up and running on the latest kernels. I have a document at https://github.com/TxBillbr/octodash-hyperpixel-fix that *might* help. It is written specifically for that screen, but works through the "black screen" problem.

UnchartedBull, 2020-08-19:
Thanks @TxBillbr for the link. Kernel 5.4 issues will be discussed in #896, there are some useful links there you might want to try @wolverine969.

closed 2020-08-19 by UnchartedBull