UnchartedBull/OctoDash #814: Fan Percentage won't be updated

nicedevil007, 2020-07-12:
I'm using marlin TFT during print to setup everything (like Z-Offset) if needed anytime.

I figured out that adjusting the fanspeed to another percentage than it was setup in the gcode won't be updated as expected in octodash.

Fan stays at the point how it is setup in the gcode.

UnchartedBull, 2020-07-13:
Yep, can't do anything about that unfortunately. Marlin doesn't report the fan speed, so OctoPrint doesn't know what the current fan speed is, thus OctoDash can't know either. DisplayLayerProgress uses the GCode to extract the fan speeds that are defined in the scripts, but that's the best info we can get, manual overwrites won't be reported anywhere ...

closed 2020-07-13 by UnchartedBull

nicedevil007, 2020-07-14:
Oh, didn't know that :/

nicedevil007, 2020-07-27:
Maybe it is fixed now? Marlin 2.0.6 :)

#define REPORT_FAN_CHANGE   // Report the new fan speed when changed by M106 (and others)

UnchartedBull, 2020-07-28:
Once OctoPrint supports that command I'll update OctoDash :) I haven't found an issue in the repo though, so maybe go ahead and open one over there :)