UnchartedBull/OctoDash #815: Sleep Mode of the Display

nicedevil007, 2020-07-12:
I checked the box in the settings to set the screen to sleep in standby mode.

Maybe I understand it wrong but I expected to see a blankscreen instead of the octodash wallpaper?
Last one will burn in the screen after some time, isn't it?

UnchartedBull, 2020-07-13:
Blanking the screen currently only works in Sleep Mode (printer is disconnected from OctoPrint). Your display also needs to support DPMS in order for this to work. 

Burn-in is only a problem with OLED screens, normal LCD panels don't suffer from that really.

nicedevil007, 2020-07-14:
> Blanking the screen currently only works in Sleep Mode (printer is disconnected from OctoPrint). Your display also needs to support DPMS in order for this to work.
> 
> Burn-in is only a problem with OLED screens, normal LCD panels don't suffer from that really.

Oh didn't know that with the burn-in only is a issue for OLEDs.

I have the original Raspberry Pi 7" screen.
If I power off the printer... I guess that is the "disconnected" status, isn't it?

nicedevil007, 2020-07-14:
so not a bug from your octodash... it is a bug of the raspberry pi 4

https://github.com/raspberrypi/linux/issues/3050

closed 2020-07-14 by nicedevil007