UnchartedBull/OctoDash #816: M600 Filamentchange

nicedevil007, 2020-07-12:
I disabled / uninstalled the filament plugin at all because I don't have that special setup with multiple color prints.
What I still want to achieve is the normal behaviour of the M600 where the nozzle goes to it's parking position and heat up pull out the filament and wait for me to say "continue" to heat up again and feed the new filament in.

If I do this the filament pull out is leading to an all time rotating extruder wheel that seems to never stop to pull out the filament.

Am I doing something wrong or is this a bug fixed in 2.0?

UnchartedBull, 2020-07-13:
It should work normally. If you disable Filament Manager you will just skip those steps automatically. Pushing out and pulling in the filament should still be available though. Is your feed length in the configuration setup correctly?

Does the progress bar move? Attaching a picture of the screen would be nice :)

nicedevil007, 2020-07-14:
To be honest... just was watching my spinning gear and that never stopped spinning... I was hypnotized :D

Will try again after my 18h print... so tomorrow.

nicedevil007, 2020-07-14:
Is working great... 18h print failed ;)

The feedlength was just ..... 470mm

closed 2020-07-14 by nicedevil007

UnchartedBull, 2020-07-14:
Damn that's a shame :(

But everything is working as expected now?