UnchartedBull/OctoDash #820: Can't enter data during the setup wizard

centauri61032, 2020-07-13:
Hey all...

Pi 3B+. with a 5" HDMI touch screen.
I have Octoprint, and Octodash installed.
Octodash starts, and tells me my new installation needs to be configured.
I hit the 'next' button, and it asks for my printer name.\
I can click the field, and it highlights. (In other words, my touch screen IS working).
However, nothing comes up that allows me to actually enter any text. 
(I was expecting some kind of pop-up on screen keyboard)
What am I doing wrong, or what do I need to configure?
Kind of a noob here, so please provide steps.

Thanks!

centauri61032, 2020-07-14:
Never mind. I just plugged a keyboard into the pi.
Thanks!

UnchartedBull, 2020-07-14:
Popup keyboard will be added soon, but isn't supported yet. Here is a workaround for now: https://github.com/UnchartedBull/OctoDash/wiki/Installation#setup-without-keyboard

closed 2020-07-14 by UnchartedBull

micheld93, 2022-12-04:
Hi, 
I installed OctoDash and it works in some way. I wonder if there is a on screen keyboard available and found this thread. What is the status of it at the moment?