UnchartedBull/OctoDash #821: WiFi not working since installed octodash

turbatus, 2020-07-15:
**Describe the bug**
Installed octodash, configured it, rebooted. After reboot no network anymore! I cannot access the rpi! On hdmi only the rainbow is shown. On the DSI LCD octodash runs fine.

**To Reproduce**
Steps to reproduce the behavior:
1. Install octodash on an running octopi (RPI3B+). Set it to run at start.
2. Reboot.
3. Configure octodash
4. Reboot
5. No network anymore. Cannot reach RPi at all!

**Expected behavior**
Octodash running with network connectivity.

**General Information:**
 - Hardware RPI3B+
 - OS Info Raspbian Buster Lite, with OctoPi]
 - OctoDash Version latest
 - OctoPrint Version latest

UnchartedBull, 2020-07-20:
OctoDash doesn't change the WiFi settings at all. Did you install additional stuff before the reboot? My best guess would be to setup WiFi again and check whether a reboot deletes the connection again. Also check your router for any errors please.

turbatus, 2020-07-20:
Well, I have tried 2 times. Install OctoDash, reboot, WiFi broken. Uninstall, reboot, Wifi works. Now uninstalled and moved to OctoScreen, for time being. Thanks anyways. Guess you can close this one.

UnchartedBull, 2020-07-20:
Mhhh that's really weird and definitely shouldn't happen. If you still want to use OctoDash - just let me know or reopen this issue, so we can have another look :)

closed 2020-07-20 by UnchartedBull

HaCo2013, 2021-01-26:
Hi,
This is the only post with the same problem that I currently have.
I made a completley fresh start (new rpi 4, installing octopi, afterwards octodash).

Witch Octoprint everything worked fine, so I moved on and installed Octodash. Didn't change anything else. After the finished install I couldn't connect to the Pi anymore. My solution right now: switching from Wifi to cable.

Any ideas why this is happening?

Thanks!
Chris

UnchartedBull, 2021-01-29:
Absolutely no idea. Have you tried removing OctoDash from the `~/.xinitrc` file and checked whether the problem still does occur?

dougalls, 2021-02-18:
Hi,

I have exactly the same issue, fresh install Octopi 0.18, wifi working fine, install Octodash and can't complete wizard as there is no wifi connection.  

I only have access to wifi currently so cannot access with a cable to edit ~/ .xinitrc through SSH.  I'll try this once I have access, is there anything else I can do in the meantime?

Thanks

Dougall

dougalls, 2021-02-19:
As an update I got hold of a keyboard, exited gui and removed octodash from ~/. xinitrc using nano and rebooted.
Following this sudo iwlist want scan returns a list including the SSID I want to connect to (and previously did connect to) but the pi won't connect to it. I tried to add SSID using raspi-config but no success. 
iwgetid does not provide any output at all.
Is there anything else I can do which would provide useful information?

dougalls, 2021-02-24:
Further update. This was a hardware issue. I was installing Octoprint ,checking it worked and then installing Octodash, shutting down and plugging in BTT PITFT50 screen before checking octodash. The screen hardware interferes with the WiFi signal on the raspberry pi 4. From another comment online from BTT I could use much larger standoffs on the screen but I am just using a USB WiFi dongle I had laying around to get over this.