UnchartedBull/OctoDash #825: Octodash not starting / only ratpoison banner

yduf, 2020-07-15:
**What doesn't work?**
the .xinitrc start ratpoison (I see the banner), but that's all.
Octodash does not show. Ratpoisoin banner stay forever

**What did you already try?**
I changed the last part of .xinitrc to the following:
```
#!/bin/bash
ratpoison&
/opt/OctoDash/octodash &> octodash.log
```

Hopping to see an error message.
octodash.log is created but is empty
.ratpoison_history is empty as well.

**General Information:**
GPIO LCD resolution set to 740x480
Linux octopi 4.19.118-v7+ #1311 SMP Mon Apr 27 14:21:24 BST 2020 armv7

- OctoDash Version 1.5.0 (master)
- OctoPrint version : 1.4.0
- OctoPi version    : 0.17.0

The same pi is able to boot successfully in desktop mode on same screen,
when changing raspi-config => B2 autologin to desktop.

Octoprint server is functionnal and can be reached.

Any idea on what I am missing ?

UnchartedBull, 2020-07-20:
Please check the troubleshooting guide: https://github.com/UnchartedBull/OctoDash/wiki/Troubleshooting#cannot-open-display-xinit-failing- and report back (especially if it is working on HDMI)

JNSmith1717, 2020-08-16:
I'm having the same issue, starts black screen witht he ratpoison at top right then goes to terminal with Xorg X Server error Illegal Instruction xinit: connection to X server lost XIO: fatal IO error 11 (Resource temporaly unavailbel) on X server ":0" after 140 requests (140 known processed) with 0 events remaining.  I have tried everything on the troubleshooting and nothing works, this is on a 5in 800x480 HDMI lcd

JNSmith1717, 2020-08-16:
also if i run ~/.xinitrc it says 
pi@octopi:~ $ ~/.xinitrc
xset:  unable to open display ":0"
xset:  unable to open display ":0"
xset:  unable to open display ":0"
Can't open display
Illegal instruction

JNSmith1717, 2020-08-16:
OctoPrint version : 1.4.2
OctoPi version    : 0.17.0
------------------------------------------------------------------------------

pi@octopi:~ $ export DISPLAY=:0
pi@octopi:~ $ ratpoison
Can't open display
pi@octopi:~ $ octodash
Illegal instruction
pi@octopi:~ $

JNSmith1717, 2020-08-16:
Using v2 octodash

UnchartedBull, 2020-08-18:
Are you sure your display is correctly connected and available on port :0. Please check if your display is running on a different number mayber: https://superuser.com/questions/647464/how-to-get-the-display-number-i-was-assigned-by-x

UnchartedBull, 2020-08-19:
Seems like you're having the asme issue as reported in #896, please comment over there if you further issues / questions.

closed 2020-08-19 by UnchartedBull

yduf, 2020-11-22:
So back to that issue => I was not able to solve it. 
I really think it was an octodash issue, since I was able to boot X11 desktop otherwise (and I had followed every possible trick listed in the FAQ).

So after some time, I decided to try to install Octodash again and see if new version solve the issue.
TLDR: it works now, that's why I am writing this:

 New version v2 didn't work on my original system, but I was able to make the same hardware work by reinstalling everything from scratch:
- Using Octopi 0.17.0
- Octoprint 1.3.12
- Octodash v2.1.2
And the hardware I am using is a [ILI9486](http://www.lcdwiki.com/MHS-3.5inch_RPi_Display),
with driver taken from [waveshare](https://github.com/waveshare/LCD-show.git)

After the standard installation the screen console work, but there is still an X11 issue, X11 does not find a display,
but this times, adding  even more packages: `sudo apt install raspberrypi-ui-mods` help solved the X11 issue.
Afterwhat Octodash start normally.

For completeness, then I encoutered issue with the touchscreen having the Y-axis inverted, this two link have helped me solved this, the trick being to use an option matrix:
- https://raspberrypi.stackexchange.com/questions/60872/inverted-gpio-touchscreen-using-99-calibration-conf
- https://medium.com/@tengfone/setting-up-raspberry-pi-4-3-5-touch-screen-xpt2046-349e484a7813