UnchartedBull/OctoDash #828: Custom buttons on printing screen

whoim2, 2020-07-16:
Hello!
Please, add in new version button to show advanced screen, have custom buttons. Need for babystep and more..

Additionaly, increase custom buttons in control screen, or place button to advanced screen from printing screen.

Thanks!

whoim2, 2020-07-21:
Initially it was about babistep, and I thought that different people may have different needs for buttons both on the print screen and on the control screen. And since there can be many of them, it is better to make them a separate screen, called by a separate button, since the 6 available buttons are usually occupied with the necessary actions and there are few of them.

> I'm not entirely sure I understand your issue here. Custom Buttons are already supported: https://github.com/UnchartedBull/OctoDash/wiki/Custom-Actions (only if the printer is idle, but OctoPrint does deny most of the commands once a print is running).
> 
> Babystepping will be tracked in #679
> 
> Am I missing something here? If so, please explain a little more detailed :)

Swap-File, 2020-08-08:
My printer is in a temperature controlled box with only a small window. My Pi running OctoDash is outside of the box.

It would be nice to be able to use a custom button to bring up the camera feed during a print to see inside.

UnchartedBull, 2020-08-10:
I'll have a look at that. Should be possible with limited scope (i.e. no axis movement). Although most of the stuff would rely on the user being somewhat resonable. I don't exactly know how OctoPrint does handle for example a homing gcode while printing. There will be no checks from OctoDash in place though ...

stenc55, 2020-09-27:
Hi. If I may drop in here, I would like to have a new empty screen with 12 (4x3 grid, fits on my 3.5 tft screen) with customizable commands, like those 6 on control screen are. I am using Klipper and there are loads of commands that needs to be done via terminal and that requires PC or tablet connection which is sometimes inconvenient or impossible ( I am not the only PC user at home :( ). I could use terminal on Octoprint, But I have not yet found a way to switch between Octodash and Octoprint.

komandrik, 2020-10-21:
During printing, there is no way to control the fan flow on the part. Can I add this? It is enough to send the M106 code with a percentage parameter.
I don't know if I need to open a separate request.

UnchartedBull, 2020-10-21:
[image]

v2.1.0 added fan speed adjustment during printing to the adjust option.

SaKiEQ, 2021-01-12:
[image]

I am having my printer within a cabinet. With OctoPrint and OctoDash + touchscreen as main terminal out side the box (lower left). An Arduino with display controls temperature etc within the box, but also lights switching. 

Now while i could simply add a physical button to the Arduino, or for that matter to Pi, it would make little sense while there is a nice GUI and touchscreen with OctoDash.

Having the ability to add a button to the main print screen, or even just while the printer is idle on the actions page would be nice. Since Arduino is communicating with Pi via Serial USB connection, it is not easy to integrate the command structure into OctoDash to call either a python script or a shell command to do so. 

While your comment above indicates that the Action Commands plugin could do so, the provided online information is way scars on that plugin. I assume I would have to send a custom GCODE M command from an OctoDash button, so that the Action Command plugin can pick it up and call the command script ?

dirkvranckaert, 2021-03-01:
Any progress on this or news on when this feature will be released?

UnchartedBull, 2021-12-01:
Hi guys! There finally is an update on this.

The next version will introduce a new notification center which is accessible from every screen (so printing included). If you pull that down you'll have access to your custom actions from everywhere. Please see the linked PR for more details and sneak peek of what this will look like.

I'm hoping to release the update soon :) In the meantime you can always build this yourself if you can't wait :)

SaKiEQ, 2021-12-01:
Beam me up Scotty !

PR #2469 (2021-12-02): Introduce Notification Center
- add a new Notification center which is accessible from every screen and houses past notification (close with press), all custom actions and a clock
- Notification center can either be opened or closed by swiping from the top to open or swiping from the bottom to close
- The icon to show the notification center is shown by default, but can be hidden if your screen supports swiping nicely

Sneak Peek:
[image]

Closes #828.
closed 2021-12-02 by UnchartedBull