UnchartedBull/OctoDash #830: Bad touchscreen work

whoim2, 2020-07-17:
Hello, im have rpi3a+ and 3,5inch spi 128mgz display with touch.

Touch work poor, see video: https://youtu.be/EZ4YfdsFXug

What problem? Thx

UnchartedBull, 2020-07-20:
Please calibrate your touchscreen and make sure that there is no residue on it. I can't and don't read the touchscreen events, because that is handled by the OS and Electron, so there is nothing I can really do here. If the screen still acts funky after calibration please try another program to see if this issue affects the other program as well, if so your screen might be broken.

closed 2020-07-20 by UnchartedBull

whoim2, 2020-07-27:
I think, rpi slow work and octodash freezes. I broke this pi, and buy 3b+, test it in 2-3 week. Tnx!

UnchartedBull, 2020-07-28:
please reopen this issue if the error occurs with the new Pi.

whoim2, 2020-08-06:
> please reopen this issue if the error occurs with the new Pi.

https://github.com/UnchartedBull/OctoDash/issues/838 last comment