UnchartedBull/OctoDash #838: Slow Touch Response

fwdixon, 2020-07-21:
**Describe the bug**
OctoDash has a slow touch response on Hyperpixel 4 with Pi 3B+. Normally have to press an option or command multiple times to get it to activate. I get the grey box indicating touch detected on each touch, so I know it is detecting the touch inputs, it just doesn't execute.

**To Reproduce**
Touch any button and see grey touch box and watch nothing happen. 

**Expected behavior**
Screen changes or command executes.

**General Information:**
 - Rapberry Pi 3B+, Hyperpixel 4.0 Touch (640X480)
 - OctoPi v0.16.0
 - OctoDash Version 1.5.0
 - OctoPrint Version 1.4.0

UnchartedBull, 2020-07-26:
Have you tried calibrating your touchscreen? Does the grey box stay there after you lift your finger from the display?

fwdixon, 2020-07-26:
The grey box does sometimes remain, but not always. 

Touchscreen is calibrated.

UnchartedBull, 2020-07-28:
That is weird. I do not have any control over the inputs, since this is all handled by electron and I just define clickable areas. Mabye try reinstalling OctoDash again? Is the same issue occurring in other apps as well?

whoim2, 2020-08-06:
> Это странно. У меня нет никакого контроля над вводом, так как все это обрабатывается электроном, и я просто определяю интерактивные области. Mabye попробовать переустановить OctoDash еще раз? Возникает ли такая же проблема и в других приложениях?

have this problem on rpi3b+ (new), https://github.com/UnchartedBull/OctoDash/issues/830
touch in desktop work normally

whoim2, 2020-08-11:
normally worked if short press with stilus

UnchartedBull, 2020-08-18:
Ok I thought I have deactivated the hold & drag feature, but I'll have a look at that again.

whoim2, 2020-08-18:
> Ok I thought I have deactivated the hold & drag feature, but I'll have a look at that again.

If need, i may test it, send me instructions and new binaries to whoim@mail.ru

greendog99, 2020-08-23:
I'm having what sounds like the same issue. I'm using an RPi4 with a 4" Miuzei 800x480 touchscreen (which has a  XPT2046 touch controller). When touching the screen I see a white box where I touch (it follows my finger around), but the button doesn't actually activate. If I touch repeatedly, eventually it works. If I use a stylus and carefully tap very quickly (i.e. to ensure I'm not "holding and dragging", it works every time. Hope this helps...

UnchartedBull, 2020-08-24:
I'll provide a fix here to test, once I get down to that :)

UnchartedBull, 2020-08-31:
Let me know if this version fixes the problem: https://drive.google.com/file/d/1QoBjetwhEF14_MhuYlmbpi2dgx30364H/view?usp=sharing

GrubbyZebra, 2020-09-01:
Getting an error saying the package is not a debian format archive.

greendog99, 2020-09-01:
> Getting an error saying the package is not a debian format archive.

Mine installed OK. However, I'm in the middle of a print so I won't be able to restart OctoPi for a couple hours.

~~~
pi@octopi:/tmp $ sudo dpkg -i ./octodash_2.0.0_armv7l_touch-fix.deb
(Reading database ... 58989 files and directories currently installed.)
Preparing to unpack .../octodash_2.0.0_armv7l_touch-fix.deb ...
Unpacking octodash (2.0.0) over (2.0.0) ...
Setting up octodash (2.0.0) ...
Processing triggers for hicolor-icon-theme (0.17-2) ...
Processing triggers for desktop-file-utils (0.23-4) ...
Processing triggers for mime-support (3.62) ...
pi@octopi:/tmp $
~~~

greendog99, 2020-09-01:
> Mine installed OK. However, I'm in the middle of a print so I won't be able to restart OctoPi for a couple hours.

I restarted OctoDash (but not the RPi) and I believe it's working better. However, I can only do a small test while it's printing. I'll report more in a few hours.

~~~
pi@octopi:/tmp $ sudo service getty@tty1 stop
pi@octopi:/tmp $ sudo systemctl daemon-reload
pi@octopi:/tmp $ sudo service getty@tty1 start
~~~

whoim2, 2020-09-01:
> Let me know if this version fixes the problem: https://drive.google.com/file/d/1QoBjetwhEF14_MhuYlmbpi2dgx30364H/view?usp=sharing
> 
> EDIT: I'll keep this PR open until one or more of you are confirming that this issue is fixed. After that the PR will be merged and the issue will be closed :)

YES!
Better work, thanks!

greendog99, 2020-09-02:
I agree, works much better. Thanks!

GrubbyZebra, 2020-09-02:
I upgraded to 2.0.0 and redownloaded, installed no problem Guess the first try was corrupted. 

This version is much slower for me. I have to hold my finger on the icon for 3-5 seconds to get it to work, whereas before it was only 1-2 seconds. Also, the overall interface has a lot more lag than the release version. 

I am getting a small round grey dot sometimes now when I press and hold, and it occasionally grows into a bigger circle, if that helps.

UnchartedBull, 2020-09-02:
@GrubbyZebra that's weird, since it seems to improve for everyone else. Are you running a lot of other stuff on the Pi? Normally the 3B+ should be able to handle OctoPrint + OctoDash without any issues. 

Are your drivers up to date? Does this issue also occur in other apps (i.e. Chrome / Desktop / ...)?

GrubbyZebra, 2020-09-02:
Just running Octoprint. no other apps installed (there is no desktop environment installed) so it is just Octoprint running on Octopi 0.16

I'll build a new pi and see how it works on that and let you know.

GrubbyZebra, 2020-09-03:
So rebuilt a new pi on a Pi3B+, nothing installed but Octopi 0.17.0, Octoprint 1.4.2, and Octodash 2.0.0 (and xdotool).

The current Octodash 2.0.0 release version is nice and responsive (so the issue I was having went away with the upgrade from 1.5.0 to 2.0.0), but the bug fix 2.0.0 version feels laggy.

Here is a short video for comparison (sorry for the shaky cam)

https://youtu.be/opgBfe4-9w4

UnchartedBull, 2020-09-07:
That definitely doesn't look right for the bugfix version, probably something due to building or some removed dependencies. I'll make sure to check performance again before the next release. Please use the official 2.0.0 build until then :)

closed 2020-09-07 by UnchartedBull

UnchartedBull, 2020-10-11:
Quick update for you guys: Problem has been solved and v2.1 which should be released soon will include the fix for the touches and have the same performance as the current official release.

GrubbyZebra, 2020-10-11:
Sweet

PR #959 (2020-09-07): Prevent more touch drag events
Fixes #838.