UnchartedBull/OctoDash #839: problems when trying to install octodash

bensaminge, 2020-07-21:
hi guys im a complete notice to coding and linux and the pi. im trying to do the automatic install for octodash and its having trouble finding some files. i will leave a copy of the text below. many thanks

bash <(wget -qO- https://github.com/UnchartedBull/OctoDash/raw/mter/scripts/install.sh)
Installing OctoDash v1.5.0
Installing Dependencies ...
1 package can be upgraded. Run 'apt list --upgradable' to see it.
Package libatspi2.0-0 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package libnss3 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package bc is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package libnotify4 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package xserver-xorg is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  xserver-xorg-core

Package desktop-file-utils is not available, but is referred to by another pacge.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package libsecret-1-0 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'libnotify4' has no installation candidate
E: Package 'libnss3' has no installation candidate
E: Package 'libatspi2.0-0' has no installation candidate
E: Unable to locate package libappindicator3-1
E: Package 'libsecret-1-0' has no installation candidate
E: Package 'xserver-xorg' has no installation candidate
E: Unable to locate package ratpoison
E: Unable to locate package xinit
E: Package 'bc' has no installation candidate
E: Package 'desktop-file-utils' has no installation candidate

fwdixon, 2020-07-21:
The auto-install script hung up for me, I had to install manually (which, btw, the instructions refer to v2.0.0, but 1.5.0 is the current release).

stevo1957, 2020-07-26:
The auto install script is not working for me either.

I am getting these errors:

Failed to fetch http://raspbian.melbourneitmirror.net/raspbian/pool/main/libe/libevdev/libevdev2_1.6.0+dfsg-1_armhf.deb

Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

stevo1957, 2020-07-26:
I just tried a manual install and am also receiving errors:

pi@octopi:~ $ sudo apt install libgtk-3-0 libnotify4 libnss3 libxss1 libxtst6 xdg-utils libatspi2.0-0 libuuid1 libappindicator3-1 libsecret-1-0 gir1.2-gnomekeyring-1.0
[sudo] password for pi:
Reading package lists... Done
Building dependency tree
Reading state information... Done
libuuid1 is already the newest version (2.33.1-0.1).
libxss1 is already the newest version (1:1.2.3-1).
libxss1 set to manually installed.
libxtst6 is already the newest version (2:1.2.3-1).
libxtst6 set to manually installed.
The following additional packages will be installed:
  adwaita-icon-theme at-spi2-core dbus-user-session dconf-gsettings-backend dconf-service gir1.2-glib-2.0 glib-networking glib-networking-common
  glib-networking-services gsettings-desktop-schemas gtk-update-icon-cache libatk-bridge2.0-0 libatk1.0-0 libatk1.0-data libauthen-sasl-perl libcairo-gobject2
  libcolord2 libcups2 libdata-dump-perl libdbusmenu-glib4 libdbusmenu-gtk3-4 libdconf1 libencode-locale-perl libepoxy0 libfile-basedir-perl libfile-desktopentry-perl
  libfile-listing-perl libfile-mimeinfo-perl libfont-afm-perl libfontenc1 libgirepository-1.0-1 libgnome-keyring-common libgnome-keyring0 libgtk-3-bin libgtk-3-common
  libhtml-form-perl libhtml-format-perl libhtml-parser-perl libhtml-tagset-perl libhtml-tree-perl libhttp-cookies-perl libhttp-daemon-perl libhttp-date-perl
  libhttp-message-perl libhttp-negotiate-perl libindicator3-7 libio-html-perl libio-socket-ssl-perl libio-stringy-perl libipc-system-simple-perl libjson-glib-1.0-0
  libjson-glib-1.0-common liblwp-mediatypes-perl liblwp-protocol-https-perl libmailtools-perl libnet-dbus-perl libnet-http-perl libnet-smtp-ssl-perl
  libnet-ssleay-perl libnspr4 libproxy1v5 librest-0.7-0 librsvg2-common libsecret-common libsoup-gnome2.4-1 libsoup2.4-1 libtie-ixhash-perl libtimedate-perl
  libtry-tiny-perl liburi-perl libwww-perl libwww-robotrules-perl libx11-protocol-perl libxaw7 libxcomposite1 libxft2 libxml-parser-perl libxml-twig-perl
  libxml-xpathengine-perl libxmu6 libxpm4 libxt6 libxxf86dga1 notification-daemon perl-openssl-defaults x11-utils x11-xserver-utils
Suggested packages:
  libdigest-hmac-perl libgssapi-perl colord cups-common gnome-keyring gvfs libcrypt-ssleay-perl libauthen-ntlm-perl libunicode-map8-perl libunicode-string-perl
  xml-twig-tools mesa-utils nickle cairo-5c xorg-docs-core
The following NEW packages will be installed:
  adwaita-icon-theme at-spi2-core dbus-user-session dconf-gsettings-backend dconf-service gir1.2-glib-2.0 gir1.2-gnomekeyring-1.0 glib-networking
  glib-networking-common glib-networking-services gsettings-desktop-schemas gtk-update-icon-cache libappindicator3-1 libatk-bridge2.0-0 libatk1.0-0 libatk1.0-data
  libatspi2.0-0 libauthen-sasl-perl libcairo-gobject2 libcolord2 libcups2 libdata-dump-perl libdbusmenu-glib4 libdbusmenu-gtk3-4 libdconf1 libencode-locale-perl
  libepoxy0 libfile-basedir-perl libfile-desktopentry-perl libfile-listing-perl libfile-mimeinfo-perl libfont-afm-perl libfontenc1 libgirepository-1.0-1
  libgnome-keyring-common libgnome-keyring0 libgtk-3-0 libgtk-3-bin libgtk-3-common libhtml-form-perl libhtml-format-perl libhtml-parser-perl libhtml-tagset-perl
  libhtml-tree-perl libhttp-cookies-perl libhttp-daemon-perl libhttp-date-perl libhttp-message-perl libhttp-negotiate-perl libindicator3-7 libio-html-perl
  libio-socket-ssl-perl libio-stringy-perl libipc-system-simple-perl libjson-glib-1.0-0 libjson-glib-1.0-common liblwp-mediatypes-perl liblwp-protocol-https-perl
  libmailtools-perl libnet-dbus-perl libnet-http-perl libnet-smtp-ssl-perl libnet-ssleay-perl libnotify4 libnspr4 libnss3 libproxy1v5 librest-0.7-0 librsvg2-common
  libsecret-1-0 libsecret-common libsoup-gnome2.4-1 libsoup2.4-1 libtie-ixhash-perl libtimedate-perl libtry-tiny-perl liburi-perl libwww-perl libwww-robotrules-perl
  libx11-protocol-perl libxaw7 libxcomposite1 libxft2 libxml-parser-perl libxml-twig-perl libxml-xpathengine-perl libxmu6 libxpm4 libxt6 libxxf86dga1
  notification-daemon perl-openssl-defaults x11-utils x11-xserver-utils xdg-utils
0 upgraded, 95 newly installed, 0 to remove and 30 not upgraded.
Need to get 25.4 MB of archives.
After this operation, 88.5 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://archive.raspberrypi.org/debian buster/main armhf libgnome-keyring-common all 3.12.0-1 [59.5 kB]
Get:2 http://mirror.aarnet.edu.au/pub/raspbian/raspbian buster/main armhf libxxf86dga1 armhf 2:1.1.4-1 [22.4 kB]
Get:4 http://archive.raspberrypi.org/debian buster/main armhf libgnome-keyring0 armhf 3.12.0-1+b1 [79.9 kB]
Get:3 http://mirror.internode.on.net/pub/raspbian/raspbian buster/main armhf librsvg2-common armhf 2.44.10-2.1+rpi1 [23.1 kB]
Get:5 http://archive.raspberrypi.org/debian buster/main armhf gtk-update-icon-cache armhf 3.24.5-1+rpt2 [80.5 kB]
Get:6 http://archive.raspberrypi.org/debian buster/main armhf gir1.2-gnomekeyring-1.0 armhf 3.12.0-1+b1 [40.4 kB]
Get:7 http://archive.raspberrypi.org/debian buster/main armhf libcairo-gobject2 armhf 1.16.0-4+rpt1 [123 kB]
Get:9 http://archive.raspberrypi.org/debian buster/main armhf libgtk-3-common all 3.24.5-1+rpt2 [3,680 kB]
Get:8 http://raspbian.melbourneitmirror.net/raspbian buster/main armhf adwaita-icon-theme all 3.30.1-1 [11.7 MB]
Get:10 http://mirror.datamossa.io/raspbian buster/main armhf libatspi2.0-0 armhf 2.30.0-7 [54.3 kB]
[... cut ...]
Get:95 http://mirror.aarnet.edu.au/pub/raspbian/raspbian buster/main armhf x11-xserver-utils armhf 7.7+8 [150 kB]
Fetched 24.2 MB in 1min 5s (374 kB/s)
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/c/cups/libcups2_2.2.10-6+deb10u1_armhf.deb  404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/n/nss/libnss3_3.42.1-1+deb10u1_armhf.deb  404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

UnchartedBull, 2020-07-26:
Could you please provide further information on your system, especially Distribution and version. I can then have a look why the packages aren't available in your repository.

@fwdixon What do you mean by hung up? Some parts of the script take a good 5 minutes on slower machines. And yeah the instructions are already updated for v2.0 which hopefully will come soon.

@stevo1957 The packages should be available. There are listed normally for buster: https://packages.debian.org/buster/libnss3. Did you try running `sudo apt update`? I've also heard of some problems with that repo, so maybe just try again :)

NightSkySK, 2020-07-27:
In my bash installations logs are as follows:
```bash <(wget -qO- https://github.com/UnchartedBull/OctoDash/raw/master/scripts/install.sh)
Installing OctoDash v1.5.0
Installing Dependencies ...
All packages are up to date.
Package gir1.2-gnomekeyring-1.0 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'gir1.2-gnomekeyring-1.0' has no installation candidate```

stevo1957, 2020-07-27:
Thanks for your help with this.

I am running an Acer laptop with Windows 10 and have previously had Octodash installed and running perfectly with a cheap 3.5" touchscreen.

I was doing a fresh install with a new Raspberry 7" touchscreen last night when I was receiving the errors.

I tried the auto install script again tonight and everything went smoothly.

I didn't do anything differently tonight except start with a fresh install of Octopi from an image backup I had, so can't explain why it worked fine tonight, but I had issues yesterday.

Thanks again for a great bit of software and for your technical support.

UnchartedBull, 2020-07-28:
I guess this was due to an error with one of your mirrors, which got fixed in between the two installations.

@NightSkySK Could you try installing again? Maybe you encountered a similar issue

closed 2020-08-19 by UnchartedBull

pciavald, 2021-01-18:
I'm getting the same issue on plain fresh debian 10 currently (manual install) : 
```
E: Package 'gir1.2-gnomekeyring-1.0' has no installation candidate
```