UnchartedBull/OctoDash #848: Black screen after running update on Pi 4

morphias2004, 2020-07-25:
**What doesn't work?**
Pi 4 with Pimoroni HyperPixel 4.0 Touch

Ran Octodash installation on OctoPrint 1.4.0 with OctoPi 0.17.0

Installed without errors and the OctoDash first run wizard ran fine.

Ran Pi APT update as I hadn't run it in a month and it loaded 153 updates.

On reboot, OctoDash no longer appears to launch. I briefly see the Pi boot on the screen, but then the screen goes black.

OctoPi is fine.

**What did you already try?**
Reinstalled latest driver from Pimoroni

Uninstalled OctoDash
Deleted .config/octodash 
Rebooted
Reinstalled OctoDash
Issue is the same and I don't even get to see the first run wizard now.
I note the config.json file does not get created.

Downloaded the sample config.json, edited it, renamed it and copied it into .config/octodash
No Change

Deleted the config.json I had copied over.

Ran sudo chmod ug+s /usr/lib/xorg/Xorg and rebooted
No Change

Double checked B2 is set in raspi-config and rebooted
No Change

Ran sudo dpkg-reconfigure x11-common and rebooted
No Change

Ran sudo apt install libpam0g-dev libx11-dev and rebooted
No Change

Uninstalled OctoDash again
Deleted .config/octodash again
Rebooted
Reinstalled OctoDash
No Change and no first run wizard

I don't have a mini HDMi to HDMi adapter to test output to an external monitor, but I do see the startup of the Pi before the screen goes black, so the screen and drivers work.

OctoDash was working prior to running the updates.

**General Information:**
Pi 4 4GB
32GB Samsung Evo Class 10 micro SD card
Latest OctoPi image from octoprint.org with OctoPrint 1.4.0 and OctoPi 0.17.0
OctoDash 1.5.0

**Additional context**
I suspect one of the plethora of updates that were loaded has broken OctoDash, but I have no idea which one.

morphias2004, 2020-07-25:
I've just done a fresh install of the Octoprint/OctoPi image, installed the Pimoroni HyperPixel drivers and installed OctoDash. Everything works correctly.

I'll now try a **sudo apt update** again and see if that breaks it again.

morphias2004, 2020-07-25:
The updates have broken it again.

Here is the list of updates:

pi@octopi:~ $ sudo apt full-upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages were automatically installed and are no longer required:
  libllvm8 rpi-eeprom-images
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed:
  busybox initramfs-tools initramfs-tools-core klibc-utils libfl2 libklibc libllvm9 linux-base pigz rpi.gpio-common
The following packages will be upgraded:
  apt apt-utils base-files bind9-host binutils binutils-arm-linux-gnueabihf binutils-common bluez bluez-firmware ca-certificates cron curl dhcpcd5
  distro-info-data dphys-swapfile e2fsprogs ffmpeg file firmware-atheros firmware-brcm80211 firmware-libertas firmware-misc-nonfree firmware-realtek
  freetype2-doc fuse git git-man imagemagick imagemagick-6-common imagemagick-6.q16 iputils-ping libapt-inst2.0 libapt-pkg5.0 libarchive13 libavcodec58
  libavdevice58 libavfilter7 libavformat58 libavresample4 libavutil56 libbind9-161 libbinutils libcom-err2 libcurl3-gnutls libcurl4 libdns-export1104
  libdns1104 libdrm-amdgpu1 libdrm-common libdrm-nouveau2 libdrm-radeon1 libdrm2 libext2fs2 libfreetype6 libfreetype6-dev libfribidi0 libfuse2 libgl1-mesa-dri
  libglib2.0-0 libglib2.0-data libgnutls30 libicu63 libidn2-0 libisc-export1100 libisc1100 libisccc161 libisccfg163 libldap-2.4-2 libldap-common liblwres161
  libmagic-mgc libmagic1 libmagickcore-6.q16-6 libmagickwand-6.q16-6 libmysofa0 libncurses6 libncursesw5 libncursesw6 libopenjp2-7 libopenmpt0
  libpam-chksshpwd libpam-modules libpam-modules-bin libpam-runtime libpam-systemd libpam0g libpango-1.0-0 libpangocairo-1.0-0 libpangoft2-1.0-0 libpostproc55
  libpython2.7 libpython2.7-dev libpython2.7-minimal libpython2.7-stdlib libpython3.7 libpython3.7-minimal libpython3.7-stdlib libraspberrypi-bin
  libraspberrypi-dev libraspberrypi-doc libraspberrypi0 libsasl2-2 libsasl2-modules-db libss2 libssh-gcrypt-4 libssl-dev libssl1.1 libswresample3 libswscale5
  libsystemd0 libtiff5 libtinfo5 libtinfo6 libudev1 libvpx5 libxml2 libxmuu1 ncurses-base ncurses-bin ncurses-term openssh-client openssh-server
  openssh-sftp-server openssl pi-bluetooth python-apt-common python-rpi.gpio python2.7 python2.7-dev python2.7-minimal python3-apt python3.7 python3.7-minimal
  raspberrypi-bootloader raspberrypi-kernel raspberrypi-sys-mods raspi-config rpcbind rpi-eeprom rpi-eeprom-images rpi-update ssh sudo systemd systemd-sysv
  tzdata udev wpasupplicant
148 upgraded, 10 newly installed, 0 to remove and 0 not upgraded.
Need to get 258 MB of archives.
After this operation, 116 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://archive.raspberrypi.org/debian buster/main armhf libpam0g armhf 1.3.1-5+rpt2 [121 kB]
Get:4 http://archive.raspberrypi.org/debian buster/main armhf libpam-modules-bin armhf 1.3.1-5+rpt2 [103 kB]
Get:5 http://archive.raspberrypi.org/debian buster/main armhf libpam-modules armhf 1.3.1-5+rpt2 [290 kB]
Get:6 http://archive.raspberrypi.org/debian buster/main armhf libpam-runtime all 1.3.1-5+rpt2 [211 kB]
Get:8 http://archive.raspberrypi.org/debian buster/main armhf bluez armhf 5.50-1.2~deb10u1+rpt1 [765 kB]
Get:10 http://archive.raspberrypi.org/debian buster/main armhf libssl-dev armhf 1.1.1d-0+deb10u3+rpt1 [1,583 kB]
Get:2 http://raspbian.melbourneitmirror.net/raspbian buster/main armhf base-files armhf 10.3+rpi1+deb10u4 [70.0 kB]
Get:3 http://raspbian.melbourneitmirror.net/raspbian buster/main armhf ncurses-bin armhf 6.1+20181013-2+deb10u2 [400 kB]
Get:7 http://raspbian.melbourneitmirror.net/raspbian buster/main armhf ncurses-base all 6.1+20181013-2+deb10u2 [258 kB]
Get:9 http://raspbian.melbourneitmirror.net/raspbian buster/main armhf systemd-sysv armhf 241-7~deb10u4+rpi1 [100 kB]
Get:11 http://raspbian.melbourneitmirror.net/raspbian buster/main armhf libpam-systemd armhf 241-7~deb10u4+rpi1 [194 kB]
Get:12 http://raspbian.melbourneitmirror.net/raspbian buster/main armhf libsystemd0 armhf 241-7~deb10u4+rpi1 [305 kB]
Get:13 http://raspbian.melbourneitmirror.net/raspbian buster/main armhf systemd armhf 241-7~deb10u4+rpi1 [3,255 kB]
Get:14 http://archive.raspberrypi.org/debian buster/main armhf libssl1.1 armhf 1.1.1d-0+deb10u3+rpt1 [1,275 kB]
Get:15 http://raspbian.melbourneitmirror.net/raspbian buster/main armhf udev armhf 241-7~deb10u4+rpi1 [1,248 kB]
Get:16 http://archive.raspberrypi.org/debian buster/main armhf bluez-firmware all 1.2-4+rpt3 [120 kB]
Get:17 http://archive.raspberrypi.org/debian buster/main armhf openssl armhf 1.1.1d-0+deb10u3+rpt1 [807 kB]
Get:18 http://raspbian.melbourneitmirror.net/raspbian buster/main armhf libudev1 armhf 241-7~deb10u4+rpi1 [144 kB]
Get:19 http://archive.raspberrypi.org/debian buster/main armhf dhcpcd5 armhf 1:8.1.2-1+rpt1 [146 kB]
Get:21 http://archive.raspberrypi.org/debian buster/main armhf dphys-swapfile all 20100506-5+rpt1 [21.4 kB]
Get:22 http://archive.raspberrypi.org/debian buster/main armhf ffmpeg armhf 7:4.1.6-1~deb10u1+rpt1 [1,432 kB]
Get:20 http://raspbian.melbourneitmirror.net/raspbian buster/main armhf libglib2.0-0 armhf 2.58.3-2+deb10u2 [1,078 kB]
Get:23 http://raspbian.melbourneitmirror.net/raspbian buster/main armhf libidn2-0 armhf 2.0.5-1+deb10u1 [74.0 kB]
Get:24 http://archive.raspberrypi.org/debian buster/main armhf libavdevice58 armhf 7:4.1.6-1~deb10u1+rpt1 [147 kB]
[... cut ...]
Get:155 http://mirror.internode.on.net/pub/raspbian/raspbian buster/main armhf rpcbind armhf 1.2.5-0.3+deb10u1 [41.2 kB]
Get:156 http://archive.raspberrypi.org/debian buster/main armhf rpi-update all 20200409 [7,488 B]
Get:157 http://mirror.internode.on.net/pub/raspbian/raspbian buster/main armhf sudo armhf 1.8.27-1+deb10u2 [1,201 kB]
Get:158 http://mirror.internode.on.net/pub/raspbian/raspbian buster/main armhf wpasupplicant armhf 2:2.7+git20190128+0c1e29f-6+deb10u2 [1,008 kB]
Fetched 258 MB in 2min 11s (1,968 kB/s)

morphias2004, 2020-07-25:
I have done another fresh install of OctoPrint/OctoPi and only loaded the HyperPixel 4 drivers. I then went straight to doing the update and full-upgrade and the issue has reoccurred.

On the up side, this conclusively rules out any fault with OctoDash.

This matter can be closed and I will open a support case on the Pimoroni HyperPixel 4 GitHub page.

https://github.com/pimoroni/hyperpixel4/issues

closed 2020-07-25 by morphias2004

morphias2004, 2020-07-25:
Case opened on Pimoroni HyperPixel 4 GitHub page:

https://github.com/pimoroni/hyperpixel4/issues/88

UnchartedBull, 2020-07-26:
Thanks for the info! 👍