UnchartedBull/OctoDash #849: Axis controls don't match OctoPrint (ie: inverted setting)

pfemiani, 2020-07-27:
When using the control screen to move the toolhead X/Y/Z the controls don't respect the axis inversion settings in OctoPrint

Steps to reproduce the behavior:
1. Go to OctoPrint sttings and choose to invert the movement of an axis.
2. In OctoPrint use the controls to move the tool head and note that the controls respect the inverted flag of any axis.
3. In OctoDash use the controls to more the tool head and note that the controls do not respect the inverted flag.

 - Raspberry Pi 3B+
 - OctoPi v0.17.0 (Raspbian GNU/Linux 10)
 - OctoDash v1.5.0
 - OctoPrint v1.4.0

UnchartedBull, 2020-07-28:
I'll try that later once I have access to the printer again. I use the API and don't send GCode directly to the printer, that's why I though OctoPrint would apply everything user configured automatically. I'll have a look at that.

pfemiani, 2020-07-29:
@UnchartedBull, I would have made the same assumption. I had some time this afternoon so I took a stab at a fix, and submitted the pull request. Fair warning, I haven't done much UI work in years and have not worked with TS or Angular before, but hopefully I wasn't too far off the mark, or at least maybe it's a starting point for you. At some point it might be worth getting the full list of printer profiles from the API and looking for the one flagged as the current profile, but for now I went ahead and just used the "_default" profile which will probably work for 99% of users.

UnchartedBull, 2020-07-29:
That's a great PR! It is definitely a very good starting point and probably will be used as is for the next couple versions until the printer profiles get fully implemented. There are some other interesting aspects in there as well, i.e. axis speeds and stuff which would make the configuration of OctoDash easier. Thank you again for your PR.

PR #856 (2020-07-30): Fix issue #849; Respect the OctoPrint axis invert flags
I went ahead and took a stab at fixing the issue where the OctoPrint axis inversion setting was not being respected in the in the printer control view.

- Created PrinterProfileService
- Implemented getDefaultPrinterProfile()
- When Control view loads it gets the default printer profile, and subsequent axis moves take the "inverted" setting into account

Fixes #849.
closed 2020-07-30 by UnchartedBull