UnchartedBull/OctoDash #850: 3.5inch driver not installing correctly

Prestat88, 2020-07-27:
Hi,

i'm an absolute novice in coding. I've been using a RPi 3 B+ with Octoprint for a while. Now i wanted to add a touchscreen with Octodash. I got it to work with TouchUI before, but with Octodash i have no luck.

I think the problem are the display drivers and not Octodash. Unfortunately i don't know anybody else to ask with this issue. So i hope it's ok to post it here.

I'm using this screen: http://www.lcdwiki.com/3.5inch_RPi_Display
(Yes you write that GPIO screens can be problematic, but i've bought it before i read your advice...)

After running "sudo ./LCD35-show" i always get this:
(Reading database ... 87869 files and directories currently installed.)
Removing xserver-xorg-input-evdev (1:2.10.3-1) ...
Processing triggers for man-db (2.8.5-2) ...
2020
need to update touch configuration
Selecting previously unselected package xserver-xorg-input-evdev.
(Reading database ... 87860 files and directories currently installed.)
Preparing to unpack .../xserver-xorg-input-evdev_1%3a2.10.6-1+b1_armhf.deb ...
Unpacking xserver-xorg-input-evdev (1:2.10.6-1+b1) ...
Setting up xserver-xorg-input-evdev (1:2.10.6-1+b1) ...
Processing triggers for man-db (2.8.5-2) ...

reboot now
Connection to octopi.local closed by remote host.
Connection to octopi.local closed.

I tried "sudo apt-get update", "sudo apt-get upgrade", "sudo apt-get dist-upgrade", etc... multiple times. I also reinstalled (or at least i tried) xserver-xorg-input-evdev and man-db. 

The guide on https://github.com/filipeflop/xpt2046-LCD-Driver-for-Raspberry-Pi tells to execute additional commands. After "sudo dpkg -i -B xserver-xorg-input-evdev_1%3a2.10.3-1_armhf.deb" i get:

dpkg: warning: downgrading xserver-xorg-input-evdev from 1:2.10.6-1+b1 to 1:2.10.3-1
(Reading database ... 87870 files and directories currently installed.)
Preparing to unpack xserver-xorg-input-evdev_1%3a2.10.3-1_armhf.deb ...
Unpacking xserver-xorg-input-evdev (1:2.10.3-1) over (1:2.10.6-1+b1) ...
dpkg: dependency problems prevent configuration of xserver-xorg-input-evdev:
 xserver-xorg-input-evdev depends on xorg-input-abi-22; however:
  Package xorg-input-abi-22 is not installed.

dpkg: error processing package xserver-xorg-input-evdev (--install):
 dependency problems - leaving unconfigured
Processing triggers for man-db (2.8.5-2) ...
Errors were encountered while processing:
 xserver-xorg-input-evdev

Maybe you can help me with my issue.

Thanks in advance!

K1W13D, 2020-07-29:
Hi,

If you are using a Chinese 3.5" LCD - that used the model number XPT2046 - give up now. They simply do not work. I spent hours on this, followed numerous threads and lost more hair that sleep.

Advice, buy a genuine 7" Raspberry Pi LCD, or one of the ore reputable brands. If you want a 3.5-4" then the HyperPixal4 is the best, if you can find one

Best of luck.....you will LOVE it once you have it running

Ray

Prestat88, 2020-07-29:
Hi Ray,

thanks for your reply. Yeah XPT2046 is exactly the one i have:/ I had hope because i managed to get TouchUI working but i gave up after hours...

I took your advice and just ordered a Hyperpixel 4.0 and a new RPi 4 :) Will the 4 perform better than the 3b+ or is Octoprint + Octodash far away from any performance issues?

Kai

K1W13D, 2020-07-29:
I’m yet to get the HyperPixal - but many proper use them , and successfully- if you don’t mind the 7” screen - they are awesome 

Keep us posted

UnchartedBull, 2020-07-30:
You shouldn't notice any huge improvements with the Pi4 as the Pi 3+ is more then sufficient already. You can't have enough power though, so good choice!

Closing this here for now thanks @K1W13D for providing the info. if you have issues with the new screen please reopen :)

closed 2020-07-30 by UnchartedBull

Prestat88, 2020-08-02:
Hi, the Hyperpixel 4.0 came and is working:) 
I'm in the setup of octodash right now and need some information. It's asking for filament feed length and feed speed. I'm using a Prusa MK3s.

I tried to google this information since 1 hour but can't find it anywhere. In Prusaslicer i also get only maximum feed speed rates.

Maybe anyone can provide me this data. 

thank you!

flukedf, 2020-12-06:
> Hi,
>
> i'm an absolute novice in coding. I've been using a RPi 3 B+ with Octoprint for a while. Now i wanted to add a touchscreen with Octodash. I got it to work with TouchUI before, but with Octodash [...]

Hello,

I had the same problem but I soldered all pins and it was working in the first test.
For configuration, I used this script in this link https://trickiknow.com/raspberry-pi-3-complete-tutorial-2018-lets-get-started/
just running like in this post and worked.
[image]
[image]
[image]